open-rdc/orne_navigation #395: α外装の要求仕様の検討

Takuyakataoka, 2019-02-28:
αの外装を作るにあたり、必要な機能や希望などを検討する。

現時点で把握している最低限の機能
・PC画面用のクリア素材の窓設置
・防水
・メンテナンス性

そのほかに何か希望等ありましたらこちらでお願いします。

yasuohayashibara, 2019-03-01:
上記の３つで構わないかと思いますが，これを満たす設計にするにはかなり大変です．
土砂降りの中でも中が濡れないようにするためには，選択できる方法は限られてしまいます．
ちなみに本体は60cm以上無いといけなかったと思いますので，現状の形状を保とうとすると必ず飛び出す部分ができてしまいます．

Takuyakataoka, 2019-03-28:
暫定的に3DCADで構想したものを作成しました

正面
[image]
裏
[image]

天板が一回り大きくなっていて傘の役割を果たす
[image]

クリア樹脂板のはまり
隙間はシリコンコーキングでふさぐ
[image]

ミーティング後の改良点
・丸みを帯びた設計にする
・シリコンからワイパーのようなゴムに付け替えられるのでは

yasuohayashibara, 2019-03-28:
まだイメージしにくいところがありますが，
以下，指摘です．

１）雨が降ったときには，風により横からも，跳ね返りにより下からも水がきます．
　　かなり下まで覆わないと，濡れるおそれがあります．
２）安全チェックで巻き込みを指摘されないか確認してください．
３）自転車と同じで車輪からの水もある程度飛び散ります．
４）サイズを大きくしたと書かれていましたが，大きくすると，その分衝突のリスクが高まります．
５）重量が重いと坂を移動できなくなることがあります．
　　ちなみに，iCart-miniのときははじめにアクリルでカバーを作りましたが，重くて坂が登れなくなりました．
６）角があると，スポンジを貼ることになります．スポンジの配置を考えないと，分解することが非常にめんどくさくなります．
７）Velodyneを搭載することを前提の設計にしたほうが良いと思います．

Takuyakataoka, 2019-04-23:
外装v2の概要図を作成

全面（黄色）と背面（白色）の2つからなる構造

＞＞黄色
・全面にPC操作のための穴（透明塩ビ板設置）
・黒いものはゴム板で外装の内側一周に設置

[image]

＞＞白色
・velodyne,URG,非常停止スイッチ用の穴
・ゴム板を外装の外側一周に設置

[image]

＜防水について＞
外装の接続部のゴム板は互い違いに設置されており水の侵入を防ぐ
図面には載っていないが接合面にゴムパッキンを付けて防水性の向上を図る

[image]

＞＞アセンブリイメージ
[image]

Takuyakataoka, 2019-04-23:
今後の検討内容
・velodyne.URG,非常停止スイッチの防水はどうするのか
（現段階の解：テーピングてカバー）
・前後の外装の接合部分に蝶番を付けて開閉式にしたらどうか
　（検討中）

必須のタスク
・USBアクセスのために黄色違法の前方を変更
・PCスペースに余裕を持たせる
　（そもそもUSBハブの固定を行えばいい可能性もあり）
・取手を移動したときの持ちはこびやすさの検討

Takuyakataoka, 2019-05-07:
FRPで何か小さいものを作ってみる

Takuyakataoka, 2019-05-11:
FRPでこちらを作成する
α外装を作成する際に必要な加工点を設けてある
[image]

Takuyakataoka, 2019-05-11:
[FRP 発注表.xlsx](https://github.com/open-rdc/orne_navigation/files/3168881/FRP.xlsx)

模型作成のため発注お願いします

yasuohayashibara, 2019-05-12:
キットではなく，個別で購入する場合，前回の残りがあると思いますので，無いものだけ購入したいと思います．
3号館3Fと新習志野にあります．新習志野のものは次回行った時に持ってくるようにします．

Akito-Shirai, 2019-08-27:
現在の状況はどうでしょうか？
更新されていないようですが、今から製作して間に合いそうですか？

Takuyakataoka, 2019-10-08:
外装の製作を考えると時間がないのでこちらで問題なければ本日中に発注表を提出します．
議論がある場合や，疑問点がある場合も10/9までにお願いします．

Takuyakataoka, 2019-10-08:
外装アセンブリイメージ
[image]
[image]

yasuohayashibara, 2019-10-09:
片岡君には昨日直接話しましたが，時間の関係もあり変更は最小限にしたほうが良いと思います．ディスプレイを搭載して配置を変える案となっていますが，無線ルータにつないでVNCでアクセスすることが現実的な解のような気がします．リモートでモニタリングした方がおそらく状況も把握しやすいと思います．

実績のある無線ルータとしては，WRH-583BK2-Sがあります．
複数接続できますので，実験走行中に各自が手元のディスプレイで挙動を確認できます．

なお，リモートで<b>操作する</b>ことは禁止ですが，<b>見るだけ</b>なら問題ありません．また，有線で繋げばスタートなども手元のコンピュータを使用して操作することができます．

あと，iPadでもVNCは使用できますので，どうしても本体にディスプレイが必要な場合は，iPadを外部に貼り付けるという方法もあります．

あと，排熱が心配な場合はファンを付けたほうが良いと思います．
雨を避けながら排熱するためには，わりと面倒な形になりますが．．．

closed 2019-11-26 by taishiyamamoto