open-rdc/orne_navigation #402: βの電装の防水と固定

JinyaIseki, 2019-05-28:


taishiyamamoto, 2019-05-28:
大まかな高層ですがシート型チューブで配線をまとめて、ケーブルグランドを使って後ろに蓋をして防水対策は回路ボックスを大きい箱に入れて二重防壁にしようかなと考えております。

シート型チューブ
https://jp.misumi-ec.com/vona2/el_wire/E1407000000/E1407010000/
ケーブルグランド
https://jp.misumi-ec.com/vona2/detail/222005037584/?KWSearch=%E3%82%BF%E3%82%AB%E3%83%81&searchFlow=results2products

taishiyamamoto, 2019-06-04:
議論の過程で底面に穴を開けたほうが良いという結論に至った。
底面に穴を開ける場合に現在の構造で穴を開けられるかどうか検討する。

daiki-shibahara, 2019-06-18:
充電回路を実装するに辺り，#342 で購入されたプラグとレセプタクルが必要な為，購入おねがいします．
D/MS3106A12S-3Sを２つ
D/MS3100A12S-3Pを２つ

yasuohayashibara, 2019-06-19:
購入する前に，まずは何をするのか明確にしてください．
コネクタの追加をするということはメカニズムの修正も意味しますので，どちらにコネクタを追加するかを明確にして，議論を得てから追加したほうが良いと思います．

ちなみに，３端子のコネクタを４つ使用するというのは計１２端子となりますが，どのように接続するかがイメージできません．

taishiyamamoto, 2019-08-06:
#416 
と一緒に防水の件を優先して取り組む

taishiyamamoto, 2019-09-17:
本走行が近づいてきたらグルーガン等で済を埋めて対処を行う

closed 2019-10-15 by JinyaIseki