open-rdc/orne_navigation #403: google cartographer で三次元地図の作成

JinyaIseki, 2019-06-06:


Akito-Shirai, 2019-06-06:
AutowareにはLoopClosureが存在しないため、三次元地図作成にはcartographerを用います。
cartographerで作成した地図をPCD形式で出力し、Autowareでナビゲーションという流れでいこうと思います。

Akito-Shirai, 2019-06-06:
cartographerでサンプルbagfileから三次元地図生成
[image]

Akito-Shirai, 2019-06-07:
２次元でLoopClosureができることを確認。↓は津田沼二号館3階。
[image]

Akito-Shirai, 2019-06-09:
現在、cartographerにて二次元のLoopClosureは確認できているが、三次元で苦戦中。
公式デモ通りにトピックを変更し、launchファイルを書き直してbagを再生したが、スキャン結果が表示されるだけで地図が生成されない。
また、同じbagfileでAutowareを用いてndt-mappingを試したところ、以下のような挙動をした。
[image]
orneのbagfile以外はndt-mappingで三次元地図を生成できているため、以下の確認を行った。
・imuが狂っているのか確かめるために、rviz上で/imu_dataを表示
　　　　⇨軽く振動している程度で特に問題なし
・orneを使わずに、Autowareでbagを取り、ndt-mappingを実行(topicは/points_rawのみ)
　　　　⇨以下のように、三次元地図(点群)を作成することができた
[image]
次に、topicを/points_rawだけでなく、/imu_data_rawも入れてみたところ、地図を作成できなかった。
以上より、個人的にはimu付近が怪しいと考えているが、今のorne環境で作成したbagfileとAutowareでは相性が悪いかもしれない。(以前はできていた)
今後、その原因追求をしていくが、今のところつくばでの三次元地図用bagfileはAutowareで取得する予定。
また、Autowareでとったbagfileをトピック名を変更してcartographerで再生してみたが、上手く行かなかった。こちらも修正していくが、もし試走会or本走行までにcartographerで三次元地図を作れなければ、LoopClosureを諦めて、Autowareで三次元地図を生成する方向も視野に入れている。

yasuohayashibara, 2019-06-10:
上記の問題に関しては，imuの向きによる影響ではないかと思われますが，推奨IMUでないことも影響しているかもしれません．
https://github.com/autowarefoundation/autoware/wiki/Overview#imu

Akito-Shirai, 2019-06-12:
cartographerで三次元地図を作ってみたが、下図のように、座標がバラバラでした。
[image]

yasuohayashibara, 2019-06-14:
こちらできるだけみなさんで協力して解決してもらえればと思います．
協力して取り組むことで，新たな知識も得られると思います．

Akito-Shirai, 2019-06-14:
Autowareで取得したbagfileで実行したところ、以下のようなエラーた出ました。
[image]
そこで、bagfile自体を検証してみたところ、以下のような結果が得られました。
[image]
ここから分かることは、最終行にて、1615ものLiDARメッセージがあるが、そのタイムスタンプが全て同値かつ0であることに問題があることです。
↓この問題と同じ状況のIssueを見つけましたが、解決に至っておりません。
https://github.com/googlecartographer/cartographer_ros/issues/540

Akito-Shirai, 2019-06-14:
正常であれば、以下のようになります。これは、デモでダウンロードしたbagfileです。
[image]

yasuohayashibara, 2019-06-17:
@Akito-Shirai 上記の地図を作成する元データを下さい．
できれば，サイズを小さくしてアップしてもらえると助かります．
不具合を確認できる（さらに不具合が解決できたことを確認できる）最低限のデータだけで構いません．

@open-rdc/project-orne 
不具合が解決できないようです．OBも含めてもし知見をお持ちの方はアドバイスいただけると助かります．

Akito-Shirai, 2019-06-17:
↓こちらのサイトでcartographerのインストールから詳細まで書かれております。
https://google-cartographer-ros.readthedocs.io/en/latest/
↓が使用したlaunch fileとlua file 、urdf です。
https://drive.google.com/drive/folders/1ZXd0A22GkPhSHwoqFPc6e7L-Tf5ydu-k?usp=sharing
↓使用したbag fileです。
https://drive.google.com/drive/folders/1ETajrEvyiPvPSRTp7DJPjYaLQq88bLBH?usp=sharing

Akito-Shirai, 2019-06-17:
大学のgoogleアカウントでないと編集できないようです。
上記ファイルを閲覧、編集する際は、大学のメールアドレスでログインするようにお願いします。

Akito-Shirai, 2019-06-17:
明日地図データ取得日ということで、上記内容よりも、優先して見て頂きたいエラーが出たため、
申し訳ありませんが下記内容を見てください。

まず初めに、orne環境でのbag fileを再生します。(bagfile⇨https://drive.google.com/open?id=1l8V5wkoKSpJbjaIWJFQvikSFGMJdFvHf)
$ roslaunch cartographer_ros demo_rslidar_3d.launch bag_filename:={full pass ~.bag}
以下の結果が得られます。
[image]
[image]

pbstreamに変換します。
①軌跡記録が終了したか確認
$ rosservice call /finish_trajectory 0
②書き込み
$ rosservice call /write_state {full pass ~.bag.pbstream}

次に、3Dbackpackのサンプル設定ファイルを使用してAssets writerを実行
$ roslaunch cartographer_ros assets_writer_backpack_3d.launch bag_filenames:={full pass ~.bag} pose_graph_filename:={full pass ~.bag.pbstream}
ここで、pbstreamからply形式に変換されるはずが、以下のエラーが出ました。
[image]

Akito-Shirai, 2019-06-17:
PCD出力までできましたが、汚い...
[image]

[image]

yasuohayashibara, 2019-06-17:
> ↓が使用したlaunch fileとlua file 、urdf です。
> https://drive.google.com/drive/folders/1ZXd0A22GkPhSHwoqFPc6e7L-Tf5ydu-k?usp=sharing
> ↓使用したbag fileです。
> https://drive.google.com/drive/folders/1ETajrEvyiPvPSRTp7DJPjYaLQq88bLBH?usp=sharing

ようやくcartographerのインストールと動作確認ができたので、上記ファイルをダウンロードしようとしたところ権限がないといわれました。権限の変更お願いします。

Akito-Shirai, 2019-06-17:
只今変更致しました。
なお、bagfileですが、「b3~」はデモ用bagfileとなっており、非常に重いです。10GBほどあります。
「autoware~はautoware」で取得したbagfileです。おそらく、cartographerで再生できないと思われます。
「2019-06-07~」は上の図と同じbagfileです。軽いので、こちらをお勧めします。

yasuohayashibara, 2019-06-17:
[b3~]はチュートリアルにありましたので、ダウンロードして地図生成できることを確認しました。
ただし、出力されているのが2Dのマップだけでしたので、3Dの点群として地図を生成できるかは確認できていません。
これからチェックします。

Akito-Shirai, 2019-06-17:
3Dは、得られた軌跡等をpbstreamに変換し、pbstreamをply形式に変換します。
その後、ply形式からpcdに変換して、三次元地図を作ります。

yasuohayashibara, 2019-06-17:
チュートリアルのTFとrqt_graph
[frames_success.pdf](https://github.com/open-rdc/orne_navigation/files/3299172/frames_success.pdf)

[image]

αで取得したデータのTFとrqt_graph
[frames.pdf](https://github.com/open-rdc/orne_navigation/files/3299173/frames.pdf)

[image]

とりあえず違うことを確認

yasuohayashibara, 2019-06-17:
データ評価の仕組みがあるようなので、とりあえず実行

```
rosrun cartographer_ros cartographer_rosbag_validate -bag_filename 2019-06-07-04-47-27.bag 
E0618 08:18:15.176692   793 rosbag_validate_main.cc:326] frame_id "imu_link" is send on multiple topics. It was seen at least on /imu/data_raw and /imu_data
E0618 08:18:15.176947   793 rosbag_validate_main.cc:326] frame_id "imu_link" is send on multiple topics. It was seen at least on /imu/data_raw and /imu_data
E0618 08:18:15.176978   793 rosbag_validate_main.cc:326] frame_id "imu_link" is send on multiple topics. It was seen at least on /imu/data_raw and /imu_data
E0618 08:18:15.178442   793 rosbag_validate_main.cc:160] Sensor with frame_id "velodyne" is not sequential in time.Previous range message ends at time 636954472474185930, current one at time 636954472474185930
E0618 08:18:15.179430   793 rosbag_validate_main.cc:160] Sensor with frame_id "velodyne" is not sequential in time.Previous range message ends at time 636954472475194290, current one at time 636954472475194290
E0618 08:18:15.180326   793 rosbag_validate_main.cc:160] Sensor with frame_id "velodyne" is not sequential in time.Previous range message ends at time 636954472476203680, current one at time 636954472476203680
E0618 08:18:15.181010   793 rosbag_validate_main.cc:160] Sensor with frame_id "velodyne" is not sequential in time.Previous range message ends at time 636954472477213250, current one at time 636954472477213250
E0618 08:18:15.181887   793 rosbag_validate_main.cc:160] Sensor with frame_id "velodyne" is not sequential in time.Previous range message ends at time 636954472478220560, current one at time 636954472478220560
E0618 08:18:15.188690   793 rosbag_validate_main.cc:160] Sensor with frame_id "velodyne" is not sequential in time.Previous range message ends at time 636954472487299180, current one at time 636954472487299180
E0618 08:18:15.231912   793 rosbag_validate_main.cc:315] Sensor with frame_id "imu_link" jumps backwards in time. Make sure that the bag contains the data for each frame_id sorted by header.stamp, i.e. the order in which they were acquired from the sensor.
E0618 08:18:15.273416   793 rosbag_validate_main.cc:315] Sensor with frame_id "imu_link" jumps backwards in time. Make sure that the bag contains the data for each frame_id sorted by header.stamp, i.e. the order in which they were acquired from the sensor.
E0618 08:18:15.293313   793 rosbag_validate_main.cc:315] Sensor with frame_id "imu_link" jumps backwards in time. Make sure that the bag contains the data for each frame_id sorted by header.stamp, i.e. the order in which they were acquired from the sensor.
W0618 08:18:16.095232   793 rosbag_validate_main.cc:203] Sensor with frame_id "velodyne" range measurements have longest overlap of 0 s
I0618 08:18:16.095702   793 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/imu/data_raw" (frame_id: "imu_link"):
Count: 23950  Min: -0.009445  Max: 0.023904  Mean: 0.005000
[-0.009445, -0.006110)	                    	Count: 3 (0.012526%)	Total: 3 (0.012526%)
[-0.006110, -0.002775)	                    	Count: 14 (0.058455%)	Total: 17 (0.070981%)
[-0.002775, 0.000560)	          ##########	Count: 11941 (49.858036%)	Total: 11958 (49.929020%)
[0.000560, 0.003894)	                    	Count: 142 (0.592902%)	Total: 12100 (50.521919%)
[0.003894, 0.007229)	                   #	Count: 1120 (4.676409%)	Total: 13220 (55.198330%)
[0.007229, 0.010564)	             #######	Count: 7790 (32.526096%)	Total: 21010 (87.724426%)
[0.010564, 0.013899)	                  ##	Count: 2172 (9.068893%)	Total: 23182 (96.793320%)
[0.013899, 0.017234)	                    	Count: 561 (2.342380%)	Total: 23743 (99.135696%)
[0.017234, 0.020569)	                    	Count: 168 (0.701461%)	Total: 23911 (99.837158%)
[0.020569, 0.023904]	                    	Count: 39 (0.162839%)	Total: 23950 (100.000000%)
I0618 08:18:16.095918   793 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/icart_mini/odom" (frame_id: "odom"):
Count: 11965  Min: 0.000340  Max: 0.026127  Mean: 0.010005
[0.000340, 0.002919)	                    	Count: 20 (0.167154%)	Total: 20 (0.167154%)
[0.002919, 0.005497)	                    	Count: 76 (0.635186%)	Total: 96 (0.802340%)
[0.005497, 0.008076)	                   #	Count: 342 (2.858337%)	Total: 438 (3.660677%)
[0.008076, 0.010655)	   #################	Count: 10347 (86.477226%)	Total: 10785 (90.137901%)
[0.010655, 0.013234)	                  ##	Count: 922 (7.705809%)	Total: 11707 (97.843712%)
[0.013234, 0.015812)	                    	Count: 196 (1.638111%)	Total: 11903 (99.481819%)
[0.015812, 0.018391)	                    	Count: 48 (0.401170%)	Total: 11951 (99.882996%)
[0.018391, 0.020970)	                    	Count: 10 (0.083577%)	Total: 11961 (99.966568%)
[0.020970, 0.023549)	                    	Count: 2 (0.016715%)	Total: 11963 (99.983284%)
[0.023549, 0.026127]	                    	Count: 2 (0.016715%)	Total: 11965 (100.000000%)
E0618 08:18:16.095949   793 rosbag_validate_main.cc:382] Point data (frame_id: "velodyne") has a large gap, largest is 0.105061 s, recommended is [0.0005, 0.05] s with no jitter.
I0618 08:18:16.096004   793 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/velodyne_points" (frame_id: "velodyne"):
Count: 2371  Min: 0.000000  Max: 0.105061  Mean: 0.050409
[0.000000, 0.010506)	          ##########	Count: 1186 (50.021088%)	Total: 1186 (50.021088%)
[... cut ...]
[0.042024, 0.052531)	                    	Count: 0 (0.000000%)	Total: 1186 (50.021088%)
[0.052531, 0.063037)	                    	Count: 0 (0.000000%)	Total: 1186 (50.021088%)
[0.063037, 0.073543)	                    	Count: 0 (0.000000%)	Total: 1186 (50.021088%)
[0.073543, 0.084049)	                    	Count: 0 (0.000000%)	Total: 1186 (50.021088%)
[0.084049, 0.094555)	                    	Count: 0 (0.000000%)	Total: 1186 (50.021088%)
[0.094555, 0.105061]	          ##########	Count: 1185 (49.978912%)	Total: 2371 (100.000000%)
```

yasuohayashibara, 2019-06-17:
チュートリアルの場合
```
rosrun cartographer_ros cartographer_rosbag_validate -bag_filename b3-2016-04-05-14-14-00.bag 
E0618 08:21:07.076095   913 rosbag_validate_main.cc:190] Sensor with frame_id "vertical_vlp16_link" sends exactly the same range measurements multiple times. Range data at time 1459865744.566555800 equals preceding data with 384 points.
E0618 08:21:07.077996   913 rosbag_validate_main.cc:190] Sensor with frame_id "vertical_vlp16_link" sends exactly the same range measurements multiple times. Range data at time 1459865744.616988000 equals preceding data with 384 points.
E0618 08:21:07.079648   913 rosbag_validate_main.cc:190] Sensor with frame_id "vertical_vlp16_link" sends exactly the same range measurements multiple times. Range data at time 1459865744.667421000 equals preceding data with 384 points.
W0618 08:21:27.283320   913 rosbag_validate_main.cc:103] frame_id imu_link time 1459866174157598600: IMU linear acceleration is 2.82385 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration 0.967735 0.141205 -2.64909
W0618 08:21:27.284610   913 rosbag_validate_main.cc:103] frame_id imu_link time 1459866174163576200: IMU linear acceleration is 2.1899 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration 0.613124 0.304309 -2.08018
W0618 08:21:27.284639   913 rosbag_validate_main.cc:103] frame_id imu_link time 1459866174163663200: IMU linear acceleration is 1.90192 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration 0.886665 0.485288 0-1.6111
I0618 08:22:07.542541   913 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "horizontal_laser_3d" (frame_id: "horizontal_vlp16_link"):
Count: 1840434  Min: 0.000026  Max: 0.014969  Mean: 0.000661
[0.000026, 0.001521)	####################	Count: 1834422 (99.673340%)	Total: 1834422 (99.673340%)
[0.001521, 0.003015)	                    	Count: 3569 (0.193922%)	Total: 1837991 (99.867264%)
[0.003015, 0.004509)	                    	Count: 1612 (0.087588%)	Total: 1839603 (99.954849%)
[0.004509, 0.006003)	                    	Count: 638 (0.034666%)	Total: 1840241 (99.989510%)
[0.006003, 0.007498)	                    	Count: 137 (0.007444%)	Total: 1840378 (99.996956%)
[0.007498, 0.008992)	                    	Count: 40 (0.002173%)	Total: 1840418 (99.999123%)
[0.008992, 0.010486)	                    	Count: 11 (0.000598%)	Total: 1840429 (99.999725%)
[0.010486, 0.011981)	                    	Count: 4 (0.000217%)	Total: 1840433 (99.999947%)
[0.011981, 0.013475)	                    	Count: 0 (0.000000%)	Total: 1840433 (99.999947%)
[0.013475, 0.014969]	                    	Count: 1 (0.000054%)	Total: 1840434 (99.999992%)
I0618 08:22:07.548790   913 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "imu" (frame_id: "imu_link"):
Count: 305307  Min: 0.000018  Max: 0.020358  Mean: 0.003999
[0.000018, 0.002052)	             #######	Count: 107007 (35.048985%)	Total: 107007 (35.048985%)
[0.002052, 0.004086)	                    	Count: 875 (0.286597%)	Total: 107882 (35.335579%)
[0.004086, 0.006120)	        ############	Count: 180670 (59.176502%)	Total: 288552 (94.512085%)
[0.006120, 0.008154)	                   #	Count: 10954 (3.587864%)	Total: 299506 (98.099945%)
[0.008154, 0.010188)	                    	Count: 748 (0.244999%)	Total: 300254 (98.344948%)
[0.010188, 0.012222)	                    	Count: 4855 (1.590203%)	Total: 305109 (99.935150%)
[0.012222, 0.014256)	                    	Count: 143 (0.046838%)	Total: 305252 (99.981987%)
[0.014256, 0.016290)	                    	Count: 33 (0.010809%)	Total: 305285 (99.992798%)
[0.016290, 0.018324)	                    	Count: 19 (0.006223%)	Total: 305304 (99.999016%)
[0.018324, 0.020358]	                    	Count: 3 (0.000983%)	Total: 305307 (100.000000%)
I0618 08:22:07.575151   913 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "vertical_laser_3d" (frame_id: "vertical_vlp16_link"):
Count: 1840434  Min: 0.000021  Max: 0.012562  Mean: 0.000662
[0.000021, 0.001275)	####################	Count: 1831750 (99.528160%)	Total: 1831750 (99.528160%)
[0.001275, 0.002530)	                    	Count: 5467 (0.297049%)	Total: 1837217 (99.825203%)
[0.002530, 0.003784)	                    	Count: 1489 (0.080905%)	Total: 1838706 (99.906105%)
[0.003784, 0.005038)	                    	Count: 1310 (0.071179%)	Total: 1840016 (99.977287%)
[0.005038, 0.006292)	                    	Count: 263 (0.014290%)	Total: 1840279 (99.991577%)
[0.006292, 0.007546)	                    	Count: 85 (0.004618%)	Total: 1840364 (99.996193%)
[0.007546, 0.008800)	                    	Count: 47 (0.002554%)	Total: 1840411 (99.998749%)
[0.008800, 0.010054)	                    	Count: 14 (0.000761%)	Total: 1840425 (99.999512%)
[0.010054, 0.011308)	                    	Count: 7 (0.000380%)	Total: 1840432 (99.999893%)
[0.011308, 0.012562]	                    	Count: 2 (0.000109%)	Total: 1840434 (99.999992%)
```

yasuohayashibara, 2019-06-17:
見る限り、/imuのトピックがないのと、タイムスタンプの問題がありそうですが、ポイントクラウドの取得周期が遅いと指摘されています。
[0.0005, 0.05] sが望ましいとのことですが、現在は最長0.1s以上かかっているとのことです。
これはデータの後処理では対処できないので、もし可能であればデータを取得する周期を短くしたほうが良いかもしれません。

yasuohayashibara, 2019-06-18:
地図生成できました．パラメータがまだ調整できていないため，あまりきれいにはできていません．
使用したファイルを自宅のPCに置忘れました．
後日アップします．

https://youtu.be/0DcS-k-Fezo

yasuohayashibara, 2019-06-18:
パラメータを調整しました。まだ調整の余地はあると思いますが、とりあえずあとはお任せします。

動画
https://youtu.be/zSLBt8Xo05s

パラメータ
[orne_alpha.zip](https://github.com/open-rdc/orne_navigation/files/3303835/orne_alpha.zip)

Akito-Shirai, 2019-06-19:
ありがとうございます。
昨日取ったbagfileで地図の作成が終わりましたら、再度アップします。

yasuohayashibara, 2019-06-20:
@Akito-Shirai 生データもどこかにアップしてもらえればと思います．

Akito-Shirai, 2019-06-24:
[image]
↑Loop Closeが上手くできません。(黒矢印の箇所)
原因として考えられるのが、
・パラメータ調整が上手くできていないこと
・bagfileを取る際にクローズさせる回数が少なかったこと
が考えられます。
使用したbagfile ⇨ https://drive.google.com/file/d/1SMHymkami4ITr456Dl4MryRmp2XsAC9p/view?usp=sharing

daiki-shibahara, 2019-07-02:
imuのセロ点調整を行う

yasuohayashibara, 2019-07-13:
オドメトリのみを使用して点群をプロットしてみました．
https://youtu.be/yphz85K3veM

以下，分かったこと

- cartographerで作成した地図と類似している．

- IMUのドリフトが見られる．
一方向にドリフトしており，これをキャンセルする後処理をすれば正確な地図が作成できるかもしれない．
次回からキャリブレーションしてからIMUを使用する．

yasuohayashibara, 2019-07-13:
gmappingを使用して地図を作成してみましたが，デフォルトのパラメータでは崩れた地図しか作成できませんでした．
現在パラメータを調整しながら作成していますが，オドメトリのみで作成した地図の方がよっぽどきちんとしています．

yasuohayashibara, 2019-07-13:
bagファイルからオドメトリのtfを抽出してプロットしてみました。
一定のドリフトをしているわけでもなさそうですので、ジャイロオドメトリのみを使用して地図を生成するのは難しそうです。
[image]

Akito-Shirai, 2019-07-14:
分かりました。
パラメータの中で、imu情報を信頼して作成する方法と信頼せずに補正をかける方法があると思いますが、後者の方が良さそうですね。
現状況としましては、上のプロット図における右下部分の確認走行エリアは綺麗にクローズできましたが、研究学園でズレてしまします。(研究学園に入ってからのz軸のズレが大きいため)
z軸のsearch_windowを拡げてみると、同じ状況になるかと思います。

yasuohayashibara, 2019-07-14:
@Akito-Shirai 
右下部分の確認走行エリアでクローズしたものがありましたら，パラメータを教えて下さい．

合計30パターンぐらい試行錯誤しましたが，私は全て一致しませんでした．
※改善はしますが，スタート付近のポールが必ず２重になります．

サンプル
https://youtu.be/fW6a_NrCem0

上記クローズしたとのことですので，対処策はあるかもしれませんが，林原としての結果が以下となります．

- このデータから地図を作成するのは難しい．

理由としては，
- ジャイロのドリフトが不定期に発生している
- 位置推定に使用するレーザレンジセンサのデータにオクルージョンが多い
ビルの壁を指標として自己位置推定したいが，その手前の障害物にレーザがあたっている
- レーザレンジセンサが振動している．また，傾けて取り付けられている．
以下の動画と比べると振動的であることが分かる．
https://www.youtube.com/watch?v=njw9BPQUoa0
さらに，やや右前が下がるような取り付けがされていると思われます．
- IMUの姿勢データとレーザレンジのデータが適切に同期していない．
旋回をしているときに，レーザの値が遅れているようすが見られる．
そのために，回転の多い部分は地図がボケたような感じになる．

rosagのデータを直接エディットすることができれば，地図を生成することができるかもしれませんが，それ以外の方法でキレイな地図を作成する方法が思いつかない状況です．

yasuohayashibara, 2019-07-14:
長くなったので，別に分けましたが，以下行ったことです．

- gmappingでの地図の作成
パラメータも少し変更してみましたが，キレイな地図を描けませんでした．

- cartographerにおいて，パラメータを様々に変更して地図作成を行った．
以下，資料
https://buildmedia.readthedocs.org/media/pdf/google-cartographer-ros/latest/google-cartographer-ros.pdf
こちらに標準的なパラメータがある．
https://github.com/googlecartographer/cartographer/tree/master/configuration_files

試した内容
- 2Dと3DでのSLAM
- レーザーレンジの距離の変更
最大を30-> 60 -> 100m
- 近くのものを無視（近くに人が常に写り込んでいたため）
最小を0 -> 1 -> 10 -> 20m
- 低いのものを無視（2m以内に動的障害物が多いため）
計測範囲 2-4m, 2-6m, 2-10m, 2-100m
- use_online_correlative_scan_matchingを使用
範囲を1 deg, 10 deg, 35 deg, 0.01 m, 0.10 m, 0.15mと変更
- pose_graph(Global SLAM)のパラメータを変更
optimize_every_n_nodesを10 -> 90 -> 350

以上を試しても理想的な地図は生成できなかった．

yasuohayashibara, 2019-07-21:
白井君が作成したluaファイルで地図を生成してみました．
ほぼ完璧な地図が描けました．

luaファイル
[190721.txt](https://github.com/open-rdc/orne_navigation/files/3414474/190721.txt)

生成した地図
https://photos.app.goo.gl/1X3eWiu1fWUTAoFQ8

生成の様子（後半１時間ぐらいは録画中を忘れて放置している動画となります．）
https://youtu.be/YBdnwmYvFXs

Akito-Shirai, 2019-07-21:
詳しくは以下に載せておきました。
https://github.com/Akito-Shirai/cartographer_ros_3d_mapping

yasuohayashibara, 2019-07-26:
cartographerを実行すると，CPUとメモリの負荷が高くなる現象ですが，林原の自宅のPCで実行して，topで状況を見たところ，以下のようになりました．

cartographer CPU 50%程度 MEM 数%程度
rviz CPU 40%程度 MEM 1.5%程度

CPUは時々150%ぐらいまで上昇することがあります．
また，MEMは徐々に上昇して，交差点ぐらいまで行くと4%程度になります．

また，nvidia-smiを実行したときの
Volatile GPU-UTILは22%程度です．

ご参考まで．

Akito-Shirai, 2019-07-28:
上記のファイルを開いてみましたが、このファイルは
~/_ws/src/cartographer_ros/cartographer_ros
の中のファイルですか？もしそのようでしたら、
~/_ws/install_isolated/share/cartographer_ros/
の中のconfiguration_fileとlaunch、urdfファイルを教えて下さい。

yasuohayashibara, 2019-07-28:
そうなります．

以下のディレクトリのファイルは変更した記憶がありません．
~/_ws/install_isolated/share/cartographer_ros/

大学に来てしまったので，確認は本日夜以降になります．

yasuohayashibara, 2019-07-29:
以下にすべてを圧縮してアップしました。
https://drive.google.com/drive/folders/1JQcohijrA9yeM1qqRDHKUinabq_68aCr?usp=sharing

Akito-Shirai, 2019-07-30:
林原先生と同じ環境で試してみましたが、以下のようにCPU,memoryが振り切り、スタックしてしまいました。
[image]

Akito-Shirai, 2019-07-30:
そこで、処理速度を上げるように書き換え、メモリがスタックした時に復帰できるように変えたところ、以下のようになりました。
[image]

Akito-Shirai, 2019-07-30:
ubuntuの環境、処理速度変更後に実行したところ、以下のように全区域クローズできました。
[image]

Akito-Shirai, 2019-07-30:
それなりの精度は出ているかと思います。
[image]

Akito-Shirai, 2019-08-06:
つくば全域でのpcd変換が厳しい為、取り敢えず三次元地図を一枚作るということで、分割して地図を作成する。

Akito-Shirai, 2019-08-20:
完成
[image]

yasuohayashibara, 2019-08-20:
作業手順を残しておいてくれると引き継ぐ人が作業しやすいので，よろしくお願いします．

Akito-Shirai, 2019-08-21:
https://github.com/Akito-Shirai/cartographer_ros_3d_mapping

Akito-Shirai, 2019-08-21:
上記リンクに作成手順、実行環境を掲載。
尚、非常に高負荷処理ですので、ノートPCや低スペックPCでは実行しないように注意してください。

closed 2019-08-21 by Akito-Shirai

yasuohayashibara, 2019-08-25:
@Akito-Shirai 

以下，自宅のPCで追試してみましたが，ループクロージャーが適正に動作しませんでした．
https://youtu.be/qGLiZehd9cg

念のためお伝えいたします．

Akito-Shirai, 2019-08-25:
ご指摘ありがとうございます。
考えられる箇所を修正してみました。
こちらでは上手く動いておりますので、確認の方をよろしくお願いします。

Akito-Shirai, 2019-08-25:
同パラメータで動かしている様子をアップしますので、少々お待ち下さい。

Akito-Shirai, 2019-08-26:
こちらのリンクに同パラメータでのデモ動画をアップしました。
https://github.com/Akito-Shirai/cartographer_ros_3d_mapping