open-rdc/orne_navigation #404: アルファのURGの接続が安定しない

JinyaIseki, 2019-06-18:


daiki-shibahara, 2019-06-18:
αやβのPCを使い自律走行を行う際，センサやモータードライバーとの接続が切れるケースが多発している．
原因として，USBの抜き差しやロボットが動作している時に発生する振動では？

以下のようなものでUSBの抜き差しを抑えられのではないか？
https://www.amazon.co.jp/Focentto-USB-%E3%82%A2%E3%83%80%E3%83%97%E3%82%BF-%E3%82%B3%E3%83%8D%E3%82%AF%E3%82%BF-2%E5%80%8B%E3%82%BB%E3%83%83%E3%83%88/dp/B071R1R9W8/ref=sxbs_sxwds-stvp?__mk_ja_JP=%E3%82%AB%E3%82%BF%E3%82%AB%E3%83%8A&keywords=usb%E3%82%BF%E3%82%A4%E3%83%97A%E3%82%AA%E3%82%B9+-+%E3%82%BF%E3%82%A4%E3%83%97A%E3%83%A1%E3%82%B9&pd_rd_i=B071R1R9W8&pd_rd_r=98456706-0f88-4769-b72b-441c80980461&pd_rd_w=FdbkT&pd_rd_wg=PitIw&pf_rd_p=e29507e3-7937-4495-82f4-55b5d5ca8047&pf_rd_r=4AY66T9RCNC3ERZ86MYW&qid=1560858259&s=computers

hoshianaaa, 2019-06-18:
αの2Dライダーは、配線を緩めたら大丈夫そうでした。

その後3階で5回自律走行して切断は見られませんでした。

yasuohayashibara, 2019-06-19:
どちらにしてもコネクタ部に振動や引張力が加わらないように，できるだけ固定するようにしてください．

hoshianaaa, 2019-06-25:
[image]

2D_LIDAR:丸くして配線に負担を減らしたら、切断しなくなった

hoshianaaa, 2019-06-25:
velodyne 緩めると電源切れるので、テープで固定した
[image]

daiki-shibahara, 2019-07-02:
電源ケーブルのコネクタを変更する
最悪の場合，velodyneの回路ボックスのコネクタと電源ケーブルを半田付けする

yasuohayashibara, 2019-07-02:
> 電源ケーブルのコネクタを変更する
> 最悪の場合，velodyneの回路ボックスのコネクタと電源ケーブルを半田付けする

上田先生に相談しながら進めて下さい．
はんだ付けなど行うと，【保証外】になると思いますので私は行わない方が良いと思います．
また，電源が入らないなど深刻な故障が発生した時に，改造したものに関しては，お金を支払っても【修理拒否】されることもあります．

きちんと対処するためには，メーカから修理の見積りを取って，大学に申請して，許可を得てから，メーカに部品を送って，修理完了するまで待つことが必要です．
（もし保証期間の場合は，無償ですので大学の許可を得る必要はありません．）

海外の製品ですので，かなり長い時間（長い場合は数ヶ月）かかるおそれがあります．
まずは上田先生から購入した業者をお聞きして，学生諸君の方から修理に関して期間も含めて業者に問い合わせたほうが良いと思います．

ryuichiueda, 2019-07-03:
こちらの代表メールアドレスに問い合わせるのがよろしいかと。ボックスがいくらかも問い合わせてくださいー

https://www.argocorp.com/

taishiyamamoto, 2019-08-06:
今の所問題が出ていないのでcloseします。

closed 2019-08-06 by taishiyamamoto