open-rdc/orne_navigation #405: アルファ用のPCでAutowareを使い自律移動を試す

JinyaIseki, 2019-06-18:


Akito-Shirai, 2019-08-06:
melodic環境でver1.12を試したところ、ビルド時にエラー。
中身を書き換えて実行して、ビルド成功したが、実行時にGUIに不具合発生。

Akito-Shirai, 2019-08-20:
作成した地図を表示
[image]

yasuohayashibara, 2019-08-20:
きれいな地図が作成できたと思います．
これはどこかの高さの点群を抽出した図でしょうか．

Akito-Shirai, 2019-08-21:
cartographerで作成したpcdファイルに、地面を０として高さ情報を与えました。

JinyaIseki, 2019-10-03:
orne-αに使用しているノートPC(ThinkPad P40 Yoga)で自律移動を試した
結果：　自己位置推定（ndt_macching）が正常に動作しなかった

動画1.1 PC操作
>https://drive.google.com/open?id=1CNSyIMvuG6GeePngMvUAdO2A6x6q0Pzm

動画1.2 走行の様子
>https://drive.google.com/open?id=1Ri1JjYf4IUBLYpbOrYOUyxJ1wqfhF4BE

参考
TIER IV ACADEMY 自動運転システム構築塾 
>3_Autoware_LP.pdf 
>>http://4c281b16296b2ab02a4e0b2e3f75446d.cdnext.stream.ne.jp/randc/mirai/3_Autoware_LP.pdf

Autoware Docs
>https://faboplatform.github.io/AutowareDocs/

場所
津田沼校舎２号館三階　エレベーターホール付近

手順

1 VLP-16のセンサデータをrosbagに記録(二回)（ThinkPad P40 Yoga）
  >3_Autoware_LP.pdf P18 19

2 三次元地図を ndt_mapping で作成(DesktopPC)
  >3_Autoware_LP.pdf P20-22

3 地図生成に使用していないrosbagで自己位置推定シミュレーション(DesktopPC)
  >3_Autoware_LP.pdf P26-30

動画2　自己位置推定シミュレーション(DesktopPC)
>https://drive.google.com/open?id=1_TePyriemeg5XrGivZly2F3tr06loOJc

4 経路生成
  >Autoware Docs >WayPoint>走行経路を生成する(DesktopPC)

5 自律走行の設定
  >Autoware Docs >Autonomous Drive>Autowareの走行設定（ThinkPad P40 Yoga）

6 Autoware からの制御信号(twist_cmd)をicart_mini/cmd_velに変更するノードを起動する（ThinkPad P40 Yoga）
　
考察

動画1.1から自己位置推定に失敗していることがわかる

確認として，手順３(動画2)をThinkPad P40 Yogaで試した結果，自己位置推定に失敗した

 動画3 自己位置推定シミュレーション（ThinkPad P40 Yoga）
 >https://drive.google.com/open?id=1lqXnFIV2vvSLz_hda311AnwJxanYj2fP

JinyaIseki, 2019-10-04:
Autoware Docs
>https://faboplatform.github.io/AutowareDocs/

Autoware Docs >Autonomous Drive>Autowareの走行設定 によると

Rvizを終了してから自律走行している（Jetson TX2）

Rvizをさせ自律走行を試す

Akito-Shirai, 2019-10-08:
ORNE環境で作成したbagfileをもとに，シミュレータ上でNDTmatchingを行った．
https://www.youtube.com/watch?v=qdEbblYM0xg

closed 2019-10-29 by JinyaIseki