open-rdc/orne_navigation #409: αとβの軸が歪んでいるため治す

daiki-shibahara, 2019-07-02:
α，β共にハの字に歪んでいる

yasuohayashibara, 2019-07-02:
写真を撮って載せたほうが良いと思います．

atsushi10mino, 2019-07-09:
正面
[image]

右車輪
[image]

左車輪
[image]

JinyaIseki, 2019-07-11:
参考
αを直進させどの程度左にずれるかの検証
条件
２ｍの直進
場所
[image]

初期位置
[image]

初期従動輪方向
[image]

結果
一回目
x:2.042
y:0.062
z:0.000

[image]

[image]

二回目
x:1.916
y:0.050
z:0.000

[image]

[image]

daiki-shibahara, 2019-07-16:
検証し、問題がなかったため閉じる

closed 2019-07-16 by daiki-shibahara