open-rdc/orne_navigation #412: 2019年度人物探索の課題に関して

taishiyamamoto, 2019-07-11:
去年と比較して以下のような変更がなされた。
-  男性、女性、子供の3パターンになった
- キャップとベストの色が走行ごとに異なる
- 反射テープがない
- 姿勢に立つ、寝る、着座などのランダムなパターンの追加

[image]
[image]
[image]

taishiyamamoto, 2019-07-11:
@yasuohayashibara 
動作実験と学習用に以下の物が必要になると思い、発注を検討しています。いかがでしょうか。
- 幼児型マネキン
- カラーベスト
- カラーキャップ

誰かつくばチャレンジで使われているマネキン等の購入元を知っていたら教えていただけると助かります。

yasuohayashibara, 2019-07-11:
安価で省スペースなものがあれば検討します．
省スペースが難しい場合は，レンタルを検討して下さい．

taishiyamamoto, 2019-07-14:
URLのリンクがホームページにありました。
男性：https://www.store-express.com/CGI/shopping/product_detail.cgi?unit_id=61-186-6
女性：https://www.store-express.com/CGI/shopping/product_detail.cgi?unit_id=61-186-5
子供：https://item.rakuten.co.jp/mznet/74126/
服：https://www.monotaro.com/g/00363632/
帽子：つくチャレ帽子 4色（緑、深緑、赤、橙）

yasuohayashibara, 2019-07-15:
次回のミーティングで議論しましょう．

taishiyamamoto, 2019-07-17:
議論の過程で下記のような方針が決まりました。

- orne-αは画像認識、orne-βは引き続き3次元点群を用いた探索を行う。

- マネキンと人を識別するためにロボットがコミュニケーションをとって、その後のリアクションの有無で判断する。

- 子供用マネキンは比較的安価で省スペースであるので購入する。また、服装はorne-αの画像認識に需要があると考えられるのでいくつか購入を検討する。

- まず、現状のアルゴリズムで反射ベストの有無、姿勢の変化にどれくらい対応できるか調査する。

よろしければこのissueはcloseして、αとβそれぞれの探索に関するissueを立てようと考えております。

yasuohayashibara, 2019-07-18:
子供のマネキンと服を２色（赤，青）購入しました．

> 子供：https://item.rakuten.co.jp/mznet/74126/
> 服：https://www.monotaro.com/g/00363632/

yasuohayashibara, 2019-07-29:
マネキン到着しました．

https://photos.app.goo.gl/wLijvepV1qf4XpeB9

taishiyamamoto, 2019-07-29:
orne_βに取り付けられている３dLIDARが現在斜めに取り付けられているがurdfモデルにそれが反映されていないようです。

https://drive.google.com/open?id=1rZw2P4LHI_Y39ted9uiZX2xhkyWA1i2y
https://drive.google.com/open?id=1M8wsrmC6VpfdS4cjkTnWVgIlBUtXM2wC

https://github.com/open-rdc/orne_navigation/blob/7f074b9c4f635024158bacc3a2db0e01b46818ea/orne_description/urdf/orne_beta.xacro#L19

taishiyamamoto, 2019-07-29:
パラメータを以下のように調整した。

https://github.com/open-rdc/orne_navigation/blob/1174f3100918a0924854df875dcaa25921bf0c18/orne_description/urdf/orne_beta.xacro#L19

taishiyamamoto, 2019-07-29:
直立した幼児型マネキンと着座した成人男性を用いて、識別精度の調査を行いました。

- 実験風景: https://drive.google.com/file/d/1lsnzXNDdRGKopkq3euZ8ZmwQsmO5JBsf/view?usp=sharing
- 動画:https://youtu.be/Kd2pbFtiTXo

また、識別の際のしきい値を下げるために、以下のようなパラメータ変更を行いました。
https://github.com/open-rdc/pcl_explore_human/commit/0858eff60f036fa4e7637775d290e4d43ed59a3e#diff-25d902c24283ab8cfbac54dfa101ad31

この人物探索のアルゴリズムは以下のようにして構成されています。

1. フィルター処理（ダウンサンプリング、地面除去）
2. クラスタリング
3. ２で得られた点群のうち、一定範囲内のサイズと反射強度を持つ点群のみをパブリッシュ
4. 3で得られた点群の特徴量を抽出し、SVMを用いて識別を行う
5. 4で人と識別されたものの座標をパブリッシュ

この調査でわかったことは

- しきい値を下げることによって反射ベストを着ていない、小さい体格の場合でも3に対応することができた。

- 着座状態かつ反射ベストをきた成人男性（好ましい条件）であっても、４に対応することができなかったため、モデルの学習を行う必要があると考えられる。

taishiyamamoto, 2019-07-29:
上の調査はLIDARの向きを修正したパラメータを用いて実験したものです。
修正前のパラメータを用いて試したところ、4で人と判断されたパターンがあることが確認できました。

taishiyamamoto, 2019-08-26:
現在のパラメータとアルゴリズムではマネキンと探索対象者の識別が困難であると判断しました。
また、学習モデルの再学習が必要になるので既存のプログラムを参考にプログラムを書き直す必要がある。
今後の方針が定まったのでこのissueはcloseします。詳細は研究ノートの方に記載しています。

closed 2019-08-26 by taishiyamamoto