open-rdc/orne_navigation #414: orne-βのimuの故障

yasuohayashibara, 2019-07-18:
@taishiyamamoto より
orne-βのimuから値が帰ってこなかったので確認してみたらピンが抜けていた．

[image]

yasuohayashibara, 2019-07-18:
以下のissueを参考にするとこれらの製品が必要になると思います。
https://github.com/open-rdc/orne_navigation/issues/371

ADIS16470/PCBZ
https://www.mouser.jp/ProductDetail/Analog-Devices/ADIS16470-PCBZ?qs=sGAEpiMZZMve4%2FbfQkoj%252BEMCJzhlHLpe%2FLS2P3%2FY4OM%3D

Devantech USB-ISS-SV USB I2C・SPI・UART変換器
https://github.com/open-rdc/orne_navigation/issues/371

yasuohayashibara, 2019-07-18:
@taishiyamamoto 
故障したのはブレイクアウトボードだけですので，こちらだけ購入できればと思います．
インターネットで簡単には見つかりませんでしたので，メーカなどに問い合わせていただければと思います．

taishiyamamoto, 2019-07-23:
アナログ・デバイセズ社のカスタマーサービスに問い合わせました。

yasuohayashibara, 2019-07-24:
@taishiyamamoto 
issueを見直してみたら，現在のIMUは以下の型番と思います．

ADIS16465-2

yasuohayashibara, 2019-07-24:
これでは？

https://www.analog.com/jp/design-center/evaluation-hardware-and-software/evaluation-boards-kits/eval-adis16imu4.html#eb-overview

taishiyamamoto, 2019-07-24:
上記の製品で間違いないと思います。
対応してくださった方々に訂正とお詫びのメールを送りました。

@yasuohayashibara 
よろしければ以下のURLからブレイクアウトボードの発注をお願いします。
https://www.mouser.jp/ProductDetail/Analog-Devices/ADIS16IMU4-PCBZ?qs=%2Fha2pyFaduhs8a%2Fi9S9r%252BKgB0sZ7P4Wd3azSNlaAieaP8vurSxAPzg==&_ga=2.210611217.526459243.1563966910-134851088.1563966910

yasuohayashibara, 2019-07-24:
発注しました．

taishiyamamoto, 2019-07-31:
部品が届いたので、故障した部品と取り替えて動作確認を行いました。
パラメータ等の変更は行っておりません。

動作確認は以下のように行いました。
https://youtu.be/Lwn5iiUh9SY

taishiyamamoto, 2019-07-31:
よろしければこのissueはcloseしてimuの配線処理を行いたいと考えております。

closed 2019-08-06 by taishiyamamoto