open-rdc/orne_navigation #418: 三次元自己位置推定

Akito-Shirai, 2019-08-27:
https://github.com/open-rdc/orne_navigation/issues/365
https://github.com/open-rdc/orne_navigation/issues/405
上記にissueに関連して、本年度orneαに三次元自己位置推定を実装する

taishiyamamoto, 2019-08-27:
https://github.com/open-rdc/orne_navigation/issues/408から派生

taishiyamamoto, 2019-10-10:
mcl_3dlを用いて屋外環境で三次元自己位置推定を行いました．
以下動画のリンクになります．
https://youtu.be/rUKVTottI0U

パラメータの調整ですが，動画ではパーティクルのサンプル数を増やしただけになっています．
現状の問題点として

- tfの依存関係を解決するために，robot_pose_ekfによる補正を行っていない．
- ヨー軸方向に視認できるほどの誤差が生まれることがあり，それの補正に時間がかかる．
- navigationパッケージの"2D Pose Estimate"で初期位置を設定できるが，amclよりも正確に位置を反映しないと間違った解に収束してしまうことがある．
- 本パッケージには膨張リセット法が実装されているが，今回の動画では用いていない．

などが挙げられます．

今後はその他のパラメータ調整を行う予定です．

taishiyamamoto, 2019-10-10:
Autowareのパッケージとmcl_3dlのパッケージを比較したとき，mcl_3dlパッケージには現状，以下のようなメリットとデメリットがあると考えられます．

メリット
- 要求される計算リソースが少ない．
- amclと互換性があるため，現在ORNEに構築されているmove_baseのシステムを使うことができる．

デメリット
- move_baseを扱う場合，Autowareと比較して単純な経路生成になる．

JinyaIseki, 2019-10-17:
ｍｃｌ-3dlを使用した走行実験を行った

走行の様子
https://drive.google.com/open?id=19w_jYgxrgILEZEZ7expW0qZFbtTsEVD_

PC画面
https://drive.google.com/open?id=1qW7OibYuWEKFP1MJhTijnhex7ObsDshF

場所
2号館3階

地図作成方法
ndt_mapping

コストマップ作成方法
openslam_gmapping

yasuohayashibara, 2019-10-18:
お疲れさまです．
だいぶできていますが，やはり自己位置が突然飛ぶ現象が見られますね．
とりあえず動くかもしれませんが，自律移動を不安定にすると思いますので，改善したほうが良いと思います．

観察して分かりましたが，現象には一定の傾向があります．
ちなみに，このような動画の解析にはKinoveaがおすすめです．

- 必ず左に回転してもとに戻る．
- 飛ぶ位置と角度は徐々に大きくなるが，短時間の中では同じ位置と姿勢である．
- レーザレンジセンサの値に基づいて突然位置が飛んでいるとは思えない．
以下の図参照

結果的に，以下が疑わしいです．
- 同じtfが２つのノードから出力されている可能性がある

odomとmapだと思われますが，rosbagを取ってグラフを書いてみると良いと思います．

- 位置が突然飛んだときのデータ
別の部分にマッチングしたとは思えない．
[image]

JinyaIseki, 2019-10-18:
frame の軌跡を確認した
odom(青) map(赤) base_link(黄)
https://drive.google.com/open?id=1bljy9fryRXP__HTyZbbnad28QKUgvvCC

tfのerrorを確認した　
`roswtf`
https://drive.google.com/open?id=1QsPlD4LUz2VZC60wIYvKEb41k-wUJu61

　base_linkに複数のtfがpublishされていることがわかる

node combine〜　がbase_linkにpublishをしないように設定した結果、エラーが解消した
https://drive.google.com/open?id=1AUrksB5ndZ_GgsgSlMRCirN57Gy0PSi6

そして走行実験を行った結果、自己位置が飛ぶ現象がなくなった
https://drive.google.com/open?id=1ZTEZOIQd1pHREV7J8XfBYj20XBBsAZnR

まだ解消していないエラーがあるが、amclを使用している環境でも同様のエラーが発生しており、
作業としての優先度は低いと思われる
https://drive.google.com/open?id=1EthhMpbUaOi2uCUGbCjs7sdRxKAoxRKG

Akito-Shirai, 2019-10-19:
津田沼校舎2号館周りにて自律走行させてみた結果，比較的自己位置は安定していた．
https://drive.google.com/file/d/1JaOs-m4E6HO9zX5GUbDzBHaw-KCqbSn0/view?usp=sharing
つくば市の市役所周りでは，津田沼校舎よりも特徴量が少ないため，おそらく同パラメータでは安定しない．

Akito-Shirai, 2019-11-05:
パラメータ調整を行い、つくば市役所周りでも自己位置を安定させた。

Akito-Shirai, 2019-11-07:
シミュレータ上でつくばチャレンジ全走行区間で完走しました．
https://youtu.be/A0v1xLVfBUU

taishiyamamoto, 2019-11-12:
完成したのでissueを閉じます。

closed 2019-11-12 by taishiyamamoto