open-rdc/orne_navigation #419: 人物探索(点群)

Akito-Shirai, 2019-08-27:
https://github.com/open-rdc/orne_navigation/issues/412
上記issueに関連して、点群を用いた人物探索を実装。

Akito-Shirai, 2019-08-27:
https://github.com/open-rdc/orne_navigation/issues/408

Akito-Shirai, 2019-09-17:
実験走行にてマネキンのbagfileを取った

taishiyamamoto, 2019-10-14:
屋内環境で実機を用いた動作実験を行った．
https://www.youtube.com/watch?v=-yaHMEaTzB0&t=1s

yasuohayashibara, 2019-10-15:
回転したりと何か動きが怪しいところがありますね．
ミーティングのときにでも何が起きているかを話していただければと思います．

taishiyamamoto, 2019-10-29:
上記の問題はorne_strategyパッケージのadd_target_searchブランチのアルゴリズムの仕様が原因でした．
探索対象者を発見した際に，発見した地点に戻って探索対象の方に向くという処理があるのですが，それの姿勢の調整が厳密なものになっているためと考えられます．
https://github.com/open-rdc/orne_navigation/blob/67fb343f8a349855044014eee35101da2881ccf2/orne_strategy/scripts/tsukuba_challenge#L214
そして，アプローチを終了したら再び発見地点の姿勢に戻るという処理を行ってから，ナビゲーションを再開します．これもまた，姿勢の調整が厳密です．
https://github.com/open-rdc/orne_navigation/blob/67fb343f8a349855044014eee35101da2881ccf2/orne_strategy/scripts/tsukuba_challenge#L247

taishiyamamoto, 2019-11-12:
校内で行った実験データをこちらにも共有してからcloseする。

taishiyamamoto, 2020-02-15:
実機で動作実験を行った結果，誤認識が多く，特に間違った方向にアプローチする現象が多々見られたため，現状の分類器ではつくばチャレンジで用いることが困難であることがわかった．

データセットの作成方法，学習スコア，環境構築の方法に関しては下記ドキュメントを参照して貰えればと思います．
http://www.hayashibara-lab.it-chiba.ac.jp/member/user/taishi/

closed 2020-02-15 by taishiyamamoto