open-rdc/orne_navigation #420: IMUとAutowareの互換性

Akito-Shirai, 2019-08-27:
https://github.com/open-rdc/orne_navigation/issues/405
上記issueの内容に関連して、orneに使われているADIS16465とAutowareの互換性を調べる。
Autoware側の推奨は9軸であるが、ADIS16465は6軸であり、orientationが足りていない。
この環境でもLocalizationできるのかを調べて貰いたい。

yasuohayashibara, 2019-08-27:
９軸というのは磁気センサを含みますでしょうか．
経験的に磁気センサは導入しないほうが自己位置推定は安定します．
念のためお伝えします．

Akito-Shirai, 2019-08-27:
はい、含みます。
分かりました。ただ、Autoware側の要求が９軸でしたので、磁気方位なしの6軸でも利用できるのか検証してみます。

Akito-Shirai, 2019-08-27:
Autoware推奨のIMUがこちらです。
https://www.analog.com/media/en/technical-documentation/data-sheets/ADIS16470.pdf

tiger0421, 2019-09-02:
[image]

autware と orne_bringupで飛ぶトピックについて。
imu/data_rawをみたところ
・autoware -> x=0.628... y=-0.335... z=-49.063...
・orne        -> x=0.116... y=-0.066... z=-9.81...

taishiyamamoto, 2019-09-02:
AutowareのGUIで立ち上げることのできるIMUのパッケージを使うとurdfが上書きされるようだったため、以下のファイルを編集しました。
/home/icart/Autoware/ros/src/sensing/drivers/imu/packages/analog_devices/launch/adis16470.launch

10行目のpublish_tfのflagをfalseに変更しました。
[image]

tiger0421, 2019-09-02:
/home/icart/Autoware/ros/src/sensing/drivers/imu/packages/analog_devices/src/adis16470.cpp
の299~300行目の値を変更

adi_driverのhistory
https://github.com/open-rdc/adi_driver/commit/3dff6168e2257ac90d24b2efe324c8e44144cd90
を参考にした

MasayaOkada, 2019-09-03:
安定するバージョンを探して環境構築する．
今の環境をどこかに残しておく．

tiger0421, 2019-09-11:
上記(https://github.com/open-rdc/orne_navigation/issues/420#issuecomment-527113688)
で変更した部分を元に戻して
~/Autoware/ros
$./catkin_make_release
をしたところエラーが出た
[image]

tiger0421, 2019-09-11:
$rostopic echo /imu/data_raw
をそれぞれ
$roslaunch orne_bringup orne_alpha.launch
と
AutowareのGUI(Runtime manager)で実行したときの様子

結果: linear_accelerationがAutoware側は5倍されている
[image]

tiger0421, 2019-09-11:
プログラムを書き換えてビルドしたところエラーログは出たものの修正箇所はビルドが通っていたみたい(?)
Autoware側での値がiCart側の値と一致した
[image]

tiger0421, 2019-10-01:
ビルドは
$catkin_make
で通るようになりました
IMUのデータはAutowareで使えそうなのでクローズします

closed 2019-10-01 by tiger0421