open-rdc/orne_navigation #423: 信号認識

shulace13, 2019-08-27:
青になった時にtopicをpublishする
トピック名は
/light_blue
としよう。
【スケジュール】
9/14でデータを集め
10月に実装

taishiyamamoto, 2019-08-27:
https://github.com/open-rdc/orne_navigation/issues/408
から派生

shulace13, 2019-09-16:
[結論]
時間的法則を発見
障害者用に、ボタンを押すと信号の時間が長くなる、というものがあったが信号時間変化は確認できなかった
全天球カメラでは歩行者信号が小さすぎてよく見えない
本来は2時間ほどの映像データが欲しかったが全天球(RICOHのTHETA)で撮影するとき
　映像データはWiFi経由でスマホのアプリに転送する
従って10分ほどのデータでさえ転送に1時間強かかったので2時間分撮影するのは現実的ではない

[法則]
午後に限ったお話であるが
つくばエクスプレスに平行方向は
青が27.98 sec
垂直方向(より混んでいる方向)
青が36.03 sec
点灯しており
両信号とも赤であった時間が7.98 sec

[今後]
つくばに行く機会を作って午前の時間帯で調査してみる(?)

[元データ]
エクセルにまとめたものをドライブか何かにあげます
ボタンを押した時の映像をYouTubeなどで共有します

shulace13, 2019-09-29:
ごめんなさいまだ生データアップロードしてなかったです、明日します。

自分用に時間制御に関するものをメモ
https://qiita.com/srs/items/a159fd8e86bb41a2f0ba

yasuohayashibara, 2019-09-30:
そういえば伝えたかどうか不安ですので，こちらに記載しておきますが，信号機の無い横断歩道は人がスタート信号を与えて良いそうです．
自動車を認識して停止しなければならないのは，信号機のある横断歩道だけとのことでした．
（自動車を認識するのはあくまでも理想で，ほとんどのロボットは信号機のみを認識してスタートすると思いますが．．．）

shulace13, 2019-10-22:
2019_10_22祝日に取得したデータ

12:27ごろ十分間でデータ取得
つくばエクスプレスに平行方向は
青が27.87 sec
垂直方向(より混んでいる方向)
青が36.00 sec
点灯しており
両信号とも赤であった時間が8.05 sec

でした．恐らくずっとこの時間なのではないかとは思いますがもう少し他の時間帯のデータもほしいところです．ノードは一応描いたので本当に時間通り制御できるかこれから検証いたします．

shulace13, 2019-10-30:
ros::Timer
が割と正確に動作しているのでこれを用いて時間制御やってみたいですね．

shulace13, 2019-10-30:
ros::Timerの問題点
nodeを立ち上げたときに、複数のタイマーを置いておくとそれぞれ同時に起動するので
node 1に対してtimer 1がよさそう。

shulace13, 2019-11-12:
切替を行うことができた
4 secごとにpublishさせています。
青は28 sec
赤は36 secの間4秒ごとに青である、赤であるをそれぞれpublishしています。
https://github.com/shulace13/signal_time/tree/devel
ここに作業していったものを置いてあります。

closed 2019-11-12 by shulace13