open-rdc/orne_navigation #428: pc充電用ケーブル購入

hoshianaaa, 2019-09-14:


hoshianaaa, 2019-09-14:
購入おねがいします

https://www.amazon.co.jp/Lenovo-ThinkPad-0B47046%E3%83%8E%E3%83%BC%E3%83%88%E3%83%91%E3%82%BD%E3%82%B3%E3%83%B3%E7%94%A8-%E5%85%85%E9%9B%BB%E3%82%B3%E3%83%B3%E3%83%90%E3%83%BC%E3%82%BF%E3%82%B1%E3%83%BC%E3%83%96%E3%83%AB-%E3%82%A2%E3%83%80%E3%83%97%E3%82%BF-%E3%83%AC%E3%83%8E%E3%83%9CThinkPad-X1%E3%82%AB%E3%83%BC%E3%83%9C%E3%83%B30B47046/dp/B01FJBGQ7U/ref=pd_bxgy_147_img_2/355-8931051-9576905?_encoding=UTF8&pd_rd_i=B01FJBGQ7U&pd_rd_r=ff24c49d-9e68-47ff-b128-6a0bd23fe6ac&pd_rd_w=bYCuY&pd_rd_wg=Lax9c&pf_rd_p=e1d64852-a4db-4f50-95af-f6bb1415e9c1&pf_rd_r=TX8KRAT25CKH7AFTVE1P&psc=1&refRID=TX8KRAT25CKH7AFTVE1P

yasuohayashibara, 2019-09-15:
購入しました．
見逃しては困るメッセージに関してはメンション @yasuohayashibara をつけましょう．

hoshianaaa, 2019-09-15:
ありがとうございます

closed 2019-09-15 by hoshianaaa