open-rdc/orne_navigation #443: 回路について

shulace13, 2019-11-13:
αやβに実装したい電流電圧計をγの方で実験を行います

前回のMTG(11/12)にてりゅうせい君の回路的に測定器に定格以上の電圧がかかるのでは？とのことだったが
γに搭載されている電源供給基板にて，12Vの供給ができそうなのでそれを用いることとなりました。

yasuohayashibara, 2019-11-14:
電源に24V以上を入力できないことは確認しましたでしょうか？
もし型番を教えてくれたら調査することができます．

yasuohayashibara, 2019-11-14:
こちらに記載されていました．
https://docs.google.com/document/d/1mCt8R_fKKehR0YAFKUCG7KNvmVjKTaYM-KgVGHul5Ic/edit#heading=h.cxch2ew1dk45

確認しました．
回路に使用しているレギュレータの定格が24Vのようですので，おそらく大丈夫なような気もしますが，安全を考えると別電源ですね．

電気回路の依存関係をできるだけ作りたくは無いとは考えていますが，仕方が無いですかね．

配線が複雑にならないようにできるだけ配慮してもらえればと思います．

shulace13, 2019-11-14:
ありがとうございます．
今回は実験も兼ねて12Vからとり，
測定器用の電源基板が出来次第そちらに移行します．

shulace13, 2019-11-14:
[image]
[image]
無事電源基板からの供給で動作確認ができました．26Vでの確認については安定化電源を用いたいと思いましたが，改めて測定器周りの回路をいじらないとだめみたいです．

また配線が結局複雑化してしまうので改めて作り直すつもりです．
バッテリの電源をひとつの基板に集め、そこから測定器と電源供給基板に分岐させるのがいいのかなと思いました．

shulace13, 2019-11-19:
αやβの回路図はdavに上がっているかもしれない
リレー回路などを用いている後もしかしたら不要かもしれないので要確認です。

根本から見直す必要あるかもしれないです。

yasuohayashibara, 2019-11-19:
αとβの回路図は夏迫さんがどこかにアップしてくれたような気がします．
もし必要であれば，直接夏迫さんに聞くことも可能です．

ryuseiotsuka, 2019-11-20:
わかりました。一度回路図をdavから探し、現在実装されている機能の把握をしたいと思います。

shulace13, 2019-11-22:
@yasuohayashibara 夏迫さんに直接きく件について、先生にお願いしてもよろしいでしょうか？

yasuohayashibara, 2019-11-23:
@shulace13 にはccしましたが，問い合わせのメールを出しました．

yasuohayashibara, 2019-12-05:
- [ ] ORNEの電子回路の回路図をアップロードする．

shulace13, 2019-12-05:
アップしましたが，このリポジトリのトップ階層にのせました．必要があれば移動をおねがいします．

yasuohayashibara, 2019-12-06:
どこ？
見つけられませんでした．
リンクを張ってもらえれば助かります．

shulace13, 2019-12-06:
https://github.com/open-rdc/orne_navigation/blob/kinetic-devel/schema.pdf
これです

yasuohayashibara, 2019-12-06:
なるほど．
レポジトリに含めるとしたら，docsの中となりますが，wikiに置いた方が良い内容です．
あと，masterに直接コミットするのを，実は禁止しています．
ブランチを作ってプルリクを投げてレビューを受けてからマージするというのが正式な方法となります．
手間をかけますが，一旦消してwikiに置いてくれると助かります．

shulace13, 2019-12-06:
すみませんでした，消しました．
ただwikiに直接画像をはれなかったのでもう少しあとになります

shulace13, 2019-12-06:
[orne_a_PowerToMotor.pdf](https://github.com/open-rdc/orne_navigation/files/3929867/orne_a_PowerToMotor.pdf)

yasuohayashibara, 2019-12-06:
こちらが適切かは分かりませんが，リンクを貼っておきました．
https://github.com/open-rdc/orne_navigation/wiki

ryuseiotsuka, 2019-12-16:
電流電圧計の定格を超えた使用ができるのかの確認を行いました
27Vまで動作を確認しました。（写真は26.5Vになっています）

taishiyamamoto, 2020-02-15:
@ryuseiotsuka @shulace13 
現在時点では電流電圧計の選定が完了したということでしょうか．
よろしければこのissueをcloseして，充電回路の設計の方をしてもらえたらと思います．

補足ですが夏迫さんが作成した遮断回路は下記issueにて廃止することを検討しています．
https://github.com/open-rdc/orne_navigation/issues/341

shulace13, 2020-02-15:
遮断回路については、次回MTGで確認したいです。
このissueは閉じても問題ないです。

充電回路の設計は別途issue建てておきますが、
αとβのドキュメント化の方を優先します。

ryuseiotsuka, 2020-02-18:
遮断回路の回路図のドキュメントは別issueかwikiに載せます。
クローズします。

closed 2020-02-18 by ryuseiotsuka