open-rdc/orne_navigation #445: トラブルシューティング

NaokiSato102, 2019-11-17:
### 初めに
Orne-γへ接続して操作しようとした際、エラーが発生し、それを解決しました。
この経験を踏まえWikiに加筆したいと考えましたが、いきなり加筆するのはどうかと思いましたのでどこに加筆すべきか、新規で項目を立ち上げるべきか、もっと一般的に論じれないか、などを協議したいと考えましたのでissueを立ち上げました。

### 本件
Orne-γへ接続して操作しようとした際、エラーが発生
#### oslaunch_icart_mini_driver_teleop_joy_launch において
```
/opt/ros/kinetic/lib/rosout/rosout: error while loading shared libraries: libroscpp.so: cannot open shared object file: No such file or directory
process[rosout-1]: started with pid [27692]
started core service [/rosout]
[rosout-1] process has died [pid 27692, exit code 127, cmd /opt/ros/kinetic/lib/rosout/rosout __name:=rosout __log:=/home/naoki/.ros/log/19c60180-05df-11ea-8362-84a6c8d02b1d/rosout-1.log].
log file: /home/naoki/.ros/log/19c60180-05df-11ea-8362-84a6c8d02b1d/rosout-1*.log
[rosout-1] restarting process
```
#### roslaunch orne_bringup orne_gamma.launch において
```
/opt/ros/kinetic/lib/rosout/rosout: error while loading shared libraries: libroscpp.so: cannot open shared object file: No such file or directory
process[rosout-1]: started with pid [26613]
started core service [/rosout]
[rosout-1] process has died [pid 26613, exit code 127, cmd /opt/ros/kinetic/lib/rosout/rosout __name:=rosout __log:=/home/naoki/.ros/log/d291be68-05dd-11ea-8362-84a6c8d02b1d/rosout-1.log].
log file: /home/naoki/.ros/log/d291be68-05dd-11ea-8362-84a6c8d02b1d/rosout-1*.log
[rosout-1] restarting process
```

#### そのほか、ROSにおいてtab補完を行おうとした際
```
rospack: error while loading shared libraries: librospack.so: cannot open shared object file: No such file or directory
rosstack: error while loading shared libraries: librospack.so: cannot open shared object file: No such file or directory
```
### 解決法
環境変数が何らかの理由で崩れていることが直接的原因。
それにより共有ライブラリなどへの正常なアクセスができていないのが原因でこのエラーが発生した。
```
error while loading shared libraries: ○○.so: cannot open shared object file: No such file or directory
```
というエラーメッセージがそれを示している。
特に、今回は**LD_LIBRARY_PATH**をいじっていたことが原因でこの事態が発生した。
### 考えうる対策
(環境変数の)定期的なバックアップ

yasuohayashibara, 2019-11-17:
他の人にとって有益と思われる情報をissueに書き込むのは良いことだと思います．
この情報は他の人も遭遇する問題かもしれないので，有益だと思います．

アドバイスとしては，

- タイトルをもっと具体的に書いて下さい．
**トラブルシューティング**ではなく**libroscpp.soをオープンできない問題**などのタイトルにすると，後で検索したときに見つけやすくなります．
- 解決法に**具体的な解決法**が書かれていない．
入力したコマンドを書いておくと，同じ問題に遭遇した人が速やかに解決できるようになり，より有益な情報になると思います．

NaokiSato102, 2019-11-18:
こんなものでよろしいですかね？
タイトルは長ったらしいですが、検索への引っかかりやすさを優先しました。

yasuohayashibara, 2019-11-18:
良くなったと思います．以下，追加の指摘です．

`error while loading shared libraries: (hogehoge).so`
で十分だと思います．長くなると一覧表示のときに２行になってしまい，表示される情報が少なくなります．
（変更しなくても構いませんが，今後のため述べておきます．）

理由：
- issue自身が「トラブルシューティング」を書くところなので，改めて書く必要がない．
- 「といったエラーが出た場合」英文で既に`error`という言葉が使用されている．

taishiyamamoto, 2019-11-19:
ミーテイングで情報を共有したのでcloseします．

closed 2019-11-19 by taishiyamamoto

NaokiSato102, 2019-11-19:
https://github.com/open-rdc/orne_navigation/wiki/ORNE%E3%81%8C%E5%8B%95%E3%81%8B%E3%81%AA%E3%81%84%E6%99%82
へ追加を決定。