open-rdc/orne_navigation #446: melodicへの移行

taishiyamamoto, 2019-11-18:
ros kineticのサポートが2021年4月にEOLされる．
現在のORNEの環境はkineticで動作しているのでmelodicへ移行する．

taishiyamamoto, 2019-11-18:
navigationパッケージをmelodicに移行するとorne_navigationが正常に動作しなくなった．
そのため，orne_nabigationの修正を行った．
修正した内容は以下の通りです．
orne_navigation:
https://github.com/taishiyamamoto/orne_navigation/tree/melodic-devel
fulanghua_navigation:
https://github.com/taishiyamamoto/fulanghua_navigation/tree/melodic-devel
icart-mini:
https://github.com/open-rdc/icart/tree/melodic-devel

Ubuntu18.04LTSのマシンで動作実験を行った．
https://www.youtube.com/watch?v=ddNPEBcc-zQ

taishiyamamoto, 2019-11-18:
動作しているが，kinetic-develのシステムを完全に移行できていない．

まだ実装されていないシステム
- [ ] 膨張リセット法
- [ ] 一部の回復行動
- [ ] 3次元自己位置推定

taishiyamamoto, 2019-11-19:
膨張リセット法を実装した．
変更内容は以下
https://github.com/taishiyamamoto/navigation/commit/39f048581b1ef3658626fbb19ebd59c540560503#diff-f9c8b2ff23a212af56d0b3de46fcf48b

hoshianaaa, 2020-01-01:
シミュレータを対応させた
https://github.com/open-rdc/icart/pull/124

taishiyamamoto, 2020-02-21:
一部の回復行動に関して，このプログラムはamclの仕様（ロボットが停止していると更新を行わない）ことが原因で実装されたものだと思うので，mcl_3dlのような逐次的に自己位置推定が行われるパッケージを用いる場合はあまり必要でない可能性があります．

yasuohayashibara, 2020-04-05:
@taishiyamamoto 
以下で動作を確認しました。
https://github.com/open-rdc/orne-box/issues/14

管理が大変ですので、open_rdcのレポジトリにもmelodicの変更を反映してもらえればと思います。

taishiyamamoto, 2020-04-09:
変更を反映するプルリクを送りました．
よろしければ確認の後，マージをお願いします．
https://github.com/open-rdc/orne_navigation/pull/463
https://github.com/open-rdc/navigation/pull/11

taishiyamamoto, 2020-04-09:
orne_navigationはwstoolを用いているため，パスをopen-rdcのものに変える必要がある．
https://github.com/taishiyamamoto/orne_navigation/blob/melodic-devel/orne_pkgs.install

yasuohayashibara, 2020-04-10:
#463 に関しては前川さんがマージしてくれたようです．
gazeboでは動作を確認したので，大丈夫だと思います．

taishiyamamoto, 2020-04-10:
コンフリクトが発生していましたが，解消しました．
https://github.com/open-rdc/navigation/pull/11

yasuohayashibara, 2020-04-11:
GitHubの操作を間違えて，こちらをいきなりプッシュしてしまいました．
すでにkineticでは検証済みで害は無いと思いますので，改めてブランチ→PRとしませんが，よろしいでしょうか．
意見のある方はお願いします．
4ad85c99e5bd440d18ee7f76370c0382e20d46b3

taishiyamamoto, 2020-04-11:
inf_is_validのフラグの内容について書かれたサイトのリンクを添付しておきます．
https://github.com/open-rdc/orne_navigation/commit/4ad85c99e5bd440d18ee7f76370c0382e20d46b3
私はプッシュで大丈夫だと思います．

taishiyamamoto, 2020-04-20:
現在，open_rdc/navigationのmelodic-develのブランチがビルドできない状況になっています．
https://github.com/open-rdc/navigation/tree/melodic-devel

taishiyamamoto, 2020-04-20:
すみません，コンフリクトを無理に解消したことが原因でビルドができなくなっていました．
@yasuohayashibara 
一旦プルリクを投げなおしてのでよろしければ確認の後マージお願いします．

yasuohayashibara, 2020-04-20:
既にPRは無いようですが解決？

taishiyamamoto, 2020-04-21:
すみません、リンクを添付します。
https://github.com/open-rdc/navigation/pull/12

tiger0421, 2020-04-21:
[image]
inf_is_validのフラグを立てたのですが/scanにはInfが混じっています．
コストマップを処理するときのみInfが変わるのでしょうか？

yasuohayashibara, 2020-04-21:
> コストマップを処理するときのみInfが変わるのでしょうか？
はいそのとおりです．

コストマップのパラメータですので，コストマップ内の処理の方法を変えます．
ローカルコストマップの生成時に，infがあると障害物が消えなくなります．
レーザレンジセンサの計測距離が環境に比べて短いと，障害物が残ったままになります．

こちらで保科さんが検証してくれています．
https://github.com/open-rdc/orne_navigation/pull/442#issuecomment-562210372

taishiyamamoto, 2020-05-19:
melodicへの移行は終わったのでcloseします．
問題が発覚した場合は別issueを立てて対応します．

closed 2020-05-19 by taishiyamamoto