open-rdc/orne_navigation #447: 今後の方針

taishiyamamoto, 2019-11-19:
#440 #441 を参考にして今後の方針を決める．

何か必要なことがあれば以下のコメント欄に記載してください．

yasuohayashibara, 2019-11-20:
提案です．

### 新しいロボットの開発 　（５年ぐらいのスパンで）

### 目的
- 学内を365日自律移動させる．（セルフ充電システム）
- 学内のお使いをさせる．（弁当の購入など）

### ロボットの仕様
- 上面フラットな台車型（マニュアルで使用できるように）
作業に応じてオプションを追加
- 人とのコミュニケーション機能	（音声認識・音声合成）
- 作業ができるように，折りたたみ式のアームもつける？

### 設計指針
- 運用しやすく
- 頑健
- 高拡張性
- 高メンテナンス性
　
### 設計に際しての工夫
- Jetson Xavier NXを使用したナビゲーション
超小型・低電力の自律移動システムの構築
- スマートフォンの利用
GPS，IMU，音声認識，合成，距離センサなどはスマートフォンの機能をできるだけ利用
https://www.itmedia.co.jp/news/articles/1812/25/news072.html

shulace13, 2019-11-26:
この台車の名前の候補募集中
・orne-box
・ORNE-QUBE

tiger0421, 2019-11-26:
UNICORNE(ユニコーン)
UNIversal and Contenalized ORNE

DaikiMaekawa, 2019-11-26:
どうもお久しぶり。 過去にORNEプロジェクトの開発に関わっていたものです。
GitHubのIssuesは定期的に拝見させていただいてます。

面白そうな議題なのでオフラインで話し合う場があれば参加しても良いですか? 弊社のオープンソースの一部も活用されているようなのでお互いに情報を共有したいです。

yasuohayashibara, 2019-11-26:
お久しぶりです．
是非ともご参加下さい．

tiger0421, 2019-12-05:
LiDARの前にアクリル板を置いたとき，アクリル板は検知されてしまう
[image]

tiger0421, 2019-12-10:
つくばチャレンジの課題，「市役所内を事前情報なしで走行する」に挑戦したい

hoshianaaa, 2019-12-23:
事前情報無し走行に役立ちそうな動画
仮想の壁をpoint cloudで作ってmove_baseのglobal_costmapに送ってる
https://www.youtube.com/watch?v=uaS009mGX_g

hoshianaaa, 2019-12-23:
作れそうなセルフ充電システム
https://www.youtube.com/watch?v=138EVW0QqXQ
赤外線センサで充電器の位置を検出してる

shulace13, 2020-01-21:
名前ってそろそろ決めちゃいますか・・・？

MasayaOkada, 2020-03-03:
## 開発方針
### 目的
α
・自律走行の完走
β
・課題達成
・研究目的

※　βで安定して動作する技術をαにも取り入れる流れで開発する

### センサ
α
・3DLiDAR
・2DURG
・IMU
・Camera(Webcamera)
β
・3DLiDAR
・2DURG
・GNSS
・Camerα(realsenseD435, WebCamera)

### 内容
α
・3次元自己位置推定
・低層障害物回避
（白線認識）
β
・3次元自己位置推定
・低層障害物回避（斜め付URG，Depth cameraなど）
・白線認識
・ER＋GR法
・信号認識
・チェックポイント，封鎖経路

closed 2020-05-26 by taishiyamamoto