open-rdc/orne_navigation #448: USBハブに関する検討

yasuohayashibara, 2019-11-25:
USBハブが原因と思われる問題が深刻であるため，このissueで情報共有及び解決したいと思います．

とりあえず，産業用のUSBハブを示します．
https://www.blackbox.co.jp/ja-jp/fi/1671/13540/USB

鉛蓄電池から電源が取れて，振動の試験もクリアしているようです．
問題はPC側のUSBの瞬断ですかね．

接点回復剤がAMAZONの評価が高いですね．
https://www.kure.com/product/detail.php?item_id=65

yasuohayashibara, 2019-11-26:
接点回復剤購入します．

yugoueda, 2019-11-26:
USB3.0対応の類似商品で価格も安いものがありましたので共有します。
https://www.startech.com/jp/Cards-Adapters/USB-3.0/Hubs/Mountable-4-Port-Rugged-Industrial-SuperSpeed-USB-3-Hub~ST4300USBM

shulace13, 2019-11-26:
締め切り
次回MTG(2019/12/10)に情報共有をしてクローズということで

NaokiSato102, 2019-11-26:
> USB3.0対応の類似商品で価格も安いものがありましたので共有します。
> https://www.startech.com/jp/Cards-Adapters/USB-3.0/Hubs/Mountable-4-Port-Rugged-Industrial-SuperSpeed-USB-3-Hub~ST4300USBM
私も同じものを見ましたが例の商品の特徴であった振動への対応の有無が（私が見つけられていないだけかもしれませんが、）不明です。

yasuohayashibara, 2019-11-27:
USBハブの候補ありがとう．
私としては，まずはサイズかなという気がしていますので，小さなサイズものがあれば紹介して下さい．

NaokiSato102, 2019-11-27:
>サイズ重視

了解しました。
最小のポート数の目安があると選定しやすいかと思います。検討お願いします。

NaokiSato102, 2019-12-10:
ミーティングより
少なくとも3ポート？
高速通信性を要するものはあまりハブ経由でない方が良い。
高負荷になるとUSBバスパワー駆動だと落ちる。外部給電必要。
ハブにハブを重ねるとあまりよくない

NaokiSato102, 2019-12-17:
USBは3.0を要求していますか？2.0まででよいですか？

[よさそうなもの](https://www.ibsjapan.co.jp/products/USH304.html)を見つけましたが如何せん値段がわかりません。

yasuohayashibara, 2019-12-17:
45,000円です．
digikeyなので，納期が早いので，ミーティングで議論してから発注します．

https://www.digikey.jp/product-detail/ja/b-b-smartworx-inc/BB-USH304/1165-1378-ND/7428806

NaokiSato102, 2019-12-17:
参考までに、検索ワードはGoogleにて「ハイリテンションポート USB」です。振動に耐性がある事を重視いたしました。

ryuseiotsuka, 2020-01-28:
物が届いたので検証は別のissueで行う。

closed 2020-01-28 by ryuseiotsuka