open-rdc/orne_navigation #450: ORNE-βに発生した問題点

taishiyamamoto, 2019-12-03:
ORNE-βの本走行時の動画とデータを確認して問題点をリストアップし，それに対する手法を検討する．

taishiyamamoto, 2019-12-03:
発生した問題点として
- 道の淵が地図に記録されていないので，waypointの配置にヒューマンエラーが生じる．
- amclだと，ロボットが停止している（速度指令が送られていない）時に観測データを破棄してしまう．
- 走行禁止領域にパーティクルがある場合それらは破棄するべきだが，それが実装されていない
- 自己位置推定の候補が複数ある場合の対処がない

hoshianaaa, 2019-12-14:
「走行禁止領域のパーティクル破棄」機能の作成
プルリクエスト[https://github.com/open-rdc/navigation/pull/10](url)

動画
破棄なし: https://www.youtube.com/watch?v=TmMtpDPNR3w&feature=youtu.be
破棄あり: https://studio.youtube.com/video/IBOvLD8lYFc/edit

hoshianaaa, 2019-12-15:
問題点 https://www.youtube.com/watch?v=QvvUqTfga5c&feature=youtu.be

hoshianaaa, 2019-12-15:
追加すべきかよくわからない

hoshianaaa, 2019-12-22:
コストのパーティクルを消すとパーティクルの重心がコストから遠ざかり、ロボットの推定位置がコストから遠ざかる.それによってコストに近づく経路生成がされ、実際のロボットはコストに入ってしまう.

hoshianaaa, 2019-12-22:
削除機能はない方が良さそうだ

closed 2020-05-26 by taishiyamamoto