open-rdc/orne_navigation #451: プラットフォームロボットの設計

yasuohayashibara, 2019-12-05:
#447 で議論したプラットフォームロボットの設計を行う．

yasuohayashibara, 2019-12-05:
まずはレーザが透過するかを検証するために，以下を購入しました．
http://www.acrysunday.co.jp/new_products/archives/cat15/cat18/post_3.html

hoshianaaa, 2019-12-06:
マルチエコー機能があるとアクリルを通過したデータを取得できる
https://www.hokuyo-aut.co.jp/products/data.php?id=52

yasuohayashibara, 2019-12-06:
透明度の高いアクリルが到着したのでそれで検証したいと思いますが，ガラスを透過することが問題になっていましたので，おそらくは透過するのではないかと考えています．
謳い文句では，ガラスより透過性が高いとなっています．
http://www.acrysunday.co.jp/products/pla_bord/01.html

tiger0421, 2019-12-17:
[image]

tiger0421, 2019-12-17:
1.5m先においたとき
[image]

shulace13, 2020-01-21:
台車には
Doog https://jp.doog-inc.com/product-thouzer.html
か
iCart-middleのどちらかがよろしいかと思います．

yasuohayashibara, 2020-02-12:
近赤外線の透過性の高いアクリルがあったが，気軽に交換できないぐらい高価です．

https://www.monotaro.com/g/04181530/

yasuohayashibara, 2020-02-17:
遅くなりましたが，プラットフォームの案を設計しました．
まだ，細部を示せていませんが，問題なければこの構成にしたいと思います．
明日のミーティングで意見を聞いて，修正した後，部品の調達を始めたいと思います．
なお，URGに関しては島田君などの実験結果から中に入れるのが難しそうと感じましたので，とりあえず外に出しています．

[image]

持ち手は折りたたんで収納しやすくしました．

[image]

[image]

yasuohayashibara, 2020-02-18:
ミーティングで特に意見が出ませんでしたので，こちらの設計で製作します．

yasuohayashibara, 2020-02-19:
設計中のデータは以下に置きます．

https://drive.google.com/drive/folders/1FTzKjHyfmug_UDPVUtk7wh9Z_zvEPqiV?usp=sharing

yasuohayashibara, 2020-02-22:
だいたいコンセプトに沿った設計が完了した．

主な追加は以下の３つです．

１）無線充電システム #460 
２）防水スピーカ　HERDIO 8541747086
３）タブレット　xperia z4 tablet

タブレットは防水・GPS・ジャイロセンサで絞り込んだら上記となった．
一応屋外でも見やすいとのレビューもあった．
タブレットは見やすい角度に調整できます．

[image]

yasuohayashibara, 2020-02-25:
現時点の設計データです．

重量　30kg
速度　1.85m (時速6.6km)
登坂能力　0.67 (33.8deg)
乗り越え可能な段差　25.8mm

yasuohayashibara, 2020-02-27:
設計が終了して，発注まで完了したためissueを閉じます．
なお，制御回路の設計が含まれていませんが，それは別issueで行います．
ちなみに，無線充電器は予算不足のため，来年度に予算があれば購入する予定です．
（まあ5年計画ですので．．．）

closed 2020-02-27 by yasuohayashibara