open-rdc/orne_navigation #454: GPSを用いたナビゲーションシステムの開発

taishiyamamoto, 2020-02-10:
GPSの座標を利用したナビゲーションシステムを開発する．

taishiyamamoto, 2020-02-10:
ZED-F9Pを搭載した以下のGNSSモジュールを購入した．
https://www.bizstation.jp/ja/drogger/dg-pro1rw_index.html

taishiyamamoto, 2020-02-10:
下記サイトを参考にwindowsでの単独測位による動作確認を行ったところ，GPS，GLONASS，Galileoから位置情報を受信できたことがわかった．
https://drogger.hatenadiary.jp/entry/dgpro1rw_windows
[image]

taishiyamamoto, 2020-02-10:
Ubuntuでの動作確認を試みたところ，Bluetoothによる接続を行ったのち，すぐに切断されてしまう問題が起こった．

taishiyamamoto, 2020-02-10:
LinuxではBlueZが標準的に使われているため，GNSSモジュールを接続できるようにBlueZのプログラムを変更する必要があると考えられる．
https://wiki.archlinux.jp/index.php/Bluetooth

taishiyamamoto, 2020-02-10:
このモジュールには拡張性を考慮してUARTピンが出ているため，有線による接続を試みる．

taishiyamamoto, 2020-02-10:
現在，コネクタとケーブルの購入を検討しています．

taishiyamamoto, 2020-02-12:
企業に相談したところ，接続しようとしているBluetoothプロトコルがsppでない可能性があるとの指摘をいただきました．

SPPとは、Bluetoothのプロファイルのひとつで、Bluetooth搭載機器を仮想のシリアルポート化するためのプロファイルのこと
https://www.kddi.com/yogo/%E6%83%85%E5%A0%B1%E3%82%B7%E3%82%B9%E3%83%86%E3%83%A0/SPP.html

らしいです．

また，LinuxにはrfcommというBluetoothシリアルポートデバイスと接続するために使うコマンドがあることがわかりました．

taishiyamamoto, 2020-02-12:
UbuntuでGNSSモジュールのBluetooth接続ができることを確認しました．
下記のシェルスクリプトを実行することで，LinuxでDG-PRO1RWをBluetooth接続することができます．
https://github.com/NaokiTakahashi12/connect-serialport-for-bluetooth
[image]

taishiyamamoto, 2020-02-12:
ROSでデータを取得できることを確認しました．
下記パッケージを用いることでNavSatFixで緯度や経度の情報をトピック通信で取得できます．
https://github.com/ros-drivers/nmea_navsat_driver

シリアルポートのパスとボーレートのみ変更してサンプルのlaunchプログラムを実行しました．

動作確認の様子の動画と得られた緯度と経度の画像を添付します．
https://youtu.be/Cx6mF2hoRpo
[image]

taishiyamamoto, 2020-02-12:
動画でも間違えていますが，シリアルポートの権限をあげないとlaunchプログラムからではシリアルポートをopenできないので
`# chmod a+rw /dev/rfcomm0`  
で変更しています．

taishiyamamoto, 2020-02-12:
モジュールの動作確認を行えたので，今後は

- [ ] 屋外環境でのデータ取り（分散などの調査）
- [ ] RTK-GNSSを行う

予定です．

taishiyamamoto, 2020-02-12:
RTK-GNSSを行う手段ですが，このモジュールにはWindowsとAndroidでアプリケーションが提供されているため，それを用いて設定等を行いたいと思います．
https://drogger.hatenadiary.jp/entry/drogger-gps_app_guide

taishiyamamoto, 2020-02-12:
17時間データを取得し続けた際の散布図を添付します．
縦軸が経度，横軸が緯度となっています．
[image]

yasuohayashibara, 2020-02-12:
経緯度ではズレが分かりにくいので，メートル単位（だいたいで構いません）とかにしてくれると助かります．

taishiyamamoto, 2020-02-15:
このサイトで測量計算すると大体55mほどの偏差があるみたいです．
屋内窓際での測量データなので，あまり参考にはならないかもしれませんが...
https://vldb.gsi.go.jp/sokuchi/surveycalc/surveycalc/bl2stf.html

yasuohayashibara, 2020-02-19:
なるほど．
屋内ならずれることが分かりましたので，今度は屋外（条件の良さそうなところ）で計測してみてください．それで自己位置推定に使用できそうなぐらいの精度がでたら，使用したいと思います．

taishiyamamoto, 2020-02-25:
条件のよさそうな屋外で1時間計測してみました．
[image]

データにはだいたい9m程度の偏差がありそうです．

yasuohayashibara, 2020-02-25:
こちらのページを使って計算すると，対角で2mでした．
https://keisan.casio.jp/exec/system/1257670779

[image]

taishiyamamoto, 2020-02-25:
すみません，計算を間違えていました．
https://vldb.gsi.go.jp/sokuchi/surveycalc/surveycalc/bl2stf.html
[image]

MasayaOkada, 2020-03-03:
データ取り内容
・条件の悪いところ
・24時間計測
・移動した軌跡

taishiyamamoto, 2020-06-30:
tree-mapsというGNSSの測位データをgooglemap上にプロットするツールを用いて軌跡を可視化しました．
https://1-dot-tree-maps-152415.appspot.com/prot/

nmea_navsat_driverを用いてデータを取得し，rosbag recordを用いてデータを記録しました．GNSSレシーバーの測位方法は単独測位です．

測位の様子を動画で撮影してないので示すことはできませんが，なるべく道の中央を歩くようにしています．

[image]

データを下記リンクにアップロードしました．
https://drive.google.com/drive/u/0/folders/1tqIuyJD9ofi97CM8PvyjpQEjVAt65B3o

taishiyamamoto, 2020-06-30:
拡大すると，ビル付近で軌跡が歪んでいることがわかります．
[image]

ryuichiueda, 2020-06-30:
でもそこまで値は飛ばないんですね

taishiyamamoto, 2020-06-30:
10~20mクラスの誤差は観測されませんでした．
憶測になるのですが，センサリセット等，補助的にセンサデータを用いるのであれば，単独測位でも運用可能かもしれないです．

ryuichiueda, 2020-06-30:
これくらいなら、誘拐からも戻りやすいかもしれません（ので、例のアルゴリズムの実装を・・・）

tiger0421, 2020-06-30:
どれくらいの周期ですか？

taishiyamamoto, 2020-06-30:
rostopic hz で確認したところ、10Hzでした。

taishiyamamoto, 2020-07-07:
モジュールの動作確認が完了したのでcloseします．今後は，壁の付近を走行した際のデータなど，様々な状況でデータを取る．（データに天気などの情報をメモすることを徹底）

closed 2020-07-07 by taishiyamamoto

luvwinnie, 2023-07-01:
@taishiyamamoto 大変参考になりました！

DG-PRO1RWSの場合でも、同じ設定で接続のでしょうか？

taishiyamamoto, 2023-07-01:
DG-PRO1RWSで試したことはありませんが、下記スクリプトのMACアドレスを変更すれば接続できると思います。

https://github.com/NaokiTakahashi12/connect-serialport-for-bluetooth/blob/master/profile/drogger/dg-pro1rw.sh#L2