open-rdc/orne_navigation #455: αとβのドキュメント化

shulace13, 2020-02-15:
αとβの回路を整理する．
本日はαの配線を確認したので回路図におこす．

また、それぞれの回路やスペックをWikiに記す．
Wikiには，現状用いられている回路の経緯を記す．
それぞれの回路を基に改良版を提案する（これについては別途issueをたてると思います）

以下ToDoまとめ

- [ ] αの回路確認

- [ ] βの回路確認

- [ ] αの回路図おこし

- [ ] βの回路図おこし

- [ ] αの情報をまとめてWikiに記す

- [ ] βの情報をまとめてWikiに記す

- [ ] αの回路を改良し検証

- [ ] βの回路を改良し検証

NaokiSato102, 2020-02-18:
γの物ですが、#445 とかもあります。

ryuseiotsuka, 2020-02-18:
wikiには情報をまとめて乗せる予定ですが、こちらのissueには情報の共有のし忘れが無いように共有すべき情報はどんどん公開していきます。

ryuseiotsuka, 2020-02-18:
ORNE-α　現在のシステム図 
[image]

ryuseiotsuka, 2020-02-18:
αの感光基板に現在の使用用途を直接書き込みました

ryuseiotsuka, 2020-02-18:
PowerSupply基板の仕様用途をまとめた
[image]
後で整理します

ryuseiotsuka, 2020-02-18:
モータードライバ基板まとめ
[image]
こちらもあとで整理します

ryuseiotsuka, 2020-03-04:
ORNE-βの回路システム図です
[image]

NaokiSato102, 2020-03-04:
実機との情報の対応の為に、整理する前の、文字を添えた写真も しっかりとした文字と囲いにクリンナップしたものがあった方がいいと思いましたが、どうでしょうか？
元画像さえいただければ数日で出来ます。

taishiyamamoto, 2020-03-10:
システムズにURGと書いてありますがORNE-βにはURGが2つ搭載されていました．
このシステムズのユニバーサルDCDCとは3D LiDARのYVT-35LXのために搭載されたものです．（今はありませんが）

shulace13, 2020-03-10:
To佐藤くん
写真は撮ってあります。
最終的にまとめる時に旧回路と新回路を載せておくといいですね。

To山本くん
指摘ありがとうございます。

To大塚Rくん
ということなので、上のシステム図に
・今は繋がれていないURGへの端子
・URGへの正しい
・3D LIDARへの端子
の三つの情報修正及び追加をお願いいたします。
でも一緒に見た時はDCDCの先にURGが繋がっていましたよね・・・？

taishiyamamoto, 2020-03-10:
すみません．そういえば途中から方針が変わっていました．[電源基板](http://t-frog.com/products/power_supply/)から[2D LiDAR](https://www.hokuyo-aut.co.jp/search/single.php?serial=21)に12Vを供給できるようになっているのですが，あまり電流値に余裕のない設計になっていました．
そのため，２つのURGから電源を供給することはできないと判断し，DCDCコンバーターからURGの電源を供給するようにしていました．(#381)
ですのでシステム図はあっています．

yasuohayashibara, 2020-03-12:
モータを遮断するリレー回路ですが，サージ吸収用のダイオードが入っていません．
これが電源回路が壊れる原因かも知れません．

使用しているリレー　LY2- DC12

shulace13, 2020-03-12:
感光基板の方ですよね。確かにβの回路周りを確認した時にも、似たような箇所が壊れていてジャンパー線を用いて修理されていましたね。

ただ今後はリレー回路を使わないで、モータ基板の電源供給にキャパシタを挟むことにしています。(#341 より)
一応の対策として、新たに設計する基板(バッテリーから3D LIDARとt-frogのPowerSupplyに分配してもらうためのもの)にグラウンドを十分に確保する、ということを検討しております。

yasuohayashibara, 2020-03-13:
以前故障したのは，電源基板TF-PW36-5/12Mになります．
以下，追える範囲で故障の履歴を示します．
今考えると，これらはリレーのサージにより故障しているおそれが高いと思います．

### 電源基板　TF-PW36-5/12M
16/01/22 故障
16/09/18 故障 これによりURGが2台連続して故障
https://github.com/open-rdc/orne_navigation/issues/201#issuecomment-299621606

ドキュメント
https://github.com/open-rdc/orne_navigation/wiki/t_frog%E9%9B%BB%E6%BA%90%E5%9F%BA%E6%9D%BF_TF_PW36_5_12M_%E6%95%85%E9%9A%9C%E8%A8%BA%E6%96%AD

### モータドライバ TF-2MD3-R6
17/05/22 初期不良？　制御基板が故障
18/04/21 ２個故障　制御基板が故障　と　ドライバ基板が発火
18/06/15 配線ミスで故障

ドキュメント
https://github.com/open-rdc/orne_navigation/wiki/t_frog%E3%83%A2%E3%83%BC%E3%82%BF%E3%83%89%E3%83%A9%E3%82%A4%E3%83%90_TF_2MD3_R6_%E6%95%85%E9%9A%9C%E8%A8%BA%E6%96%AD

### リレーのサージ
制御回路系を製作する方　@ryuseiotsuka ?

参考までに以前リレーのサージを計測したときの様子を示します．
（別の回路です．）
電源電圧の数倍の電圧が一時的に発生して，電子回路を破壊します．

ということで，リレーをON/OFFするときの，12V電源の波形を確認したほうが良いです．
オシロスコープで確認してもらえればと思います．

ちなみに，壊れたURGは修理不可能でそのまま廃棄となりました．
挽回は難しいので，電源はよくチェックして製作していただければと思います．

[image]

taishiyamamoto, 2020-03-17:
ORNE-αを用いてどれくらいサージが発生しているのかチェックする．

ryuseiotsuka, 2020-05-19:
α・βのシステム図に誤りがある可能性があるので確認する必要がある。

taishiyamamoto, 2020-06-16:
graph_navigationのパッケージ
https://github.com/Norihiro02/graph_navi

ryuseiotsuka, 2020-06-25:
サージ電流計測結果
ボタンを押す動作（電流を遮断する動作）がかなり回路の負担になっていることが分かった。
電源遮断時
[image]
電源復旧時
[image]

taishiyamamoto, 2020-07-20:
現在のαとβとγのシステム図

**ORNE-α**
[image]

**ORNE-β**
[image]

**ORNE-γ**
[image]

ryuseiotsuka, 2020-07-20:
システム図ありがとうございます
ToDOリストのα,βの回路図おこしなのですが、回路の大半が既製品なのでチェックをつけていいと思うのですがどうでしょうか。

ryuseiotsuka, 2020-07-21:
α,βの回路図おこしは上記のシステム図で達成とする

taishiyamamoto, 2020-07-22:
ORNEの回路に関してまとめたものをwikiに書きました。
https://github.com/open-rdc/orne_navigation/wiki/ORNE%E3%81%AE%E5%9B%9E%E8%B7%AF%E5%9B%B3

taishiyamamoto, 2020-07-22:
@ryuseiotsuka 
回路の改良と検証は#482 など、別途issueを立てて行っているので閉じても大丈夫かなと思っています。

taishiyamamoto, 2020-07-28:
ドキュメントの作成が完了したので閉じる．

closed 2020-07-28 by taishiyamamoto