open-rdc/orne_navigation #458: αの外装の開発

tiger0421, 2020-02-18:
### 目的
つくば  

### 要求
- 防水
- 配線
- 重量

### Sensors
- IMU(ADIS16465)
- Rfans
- GPS
- web camera or Realsense

yasuohayashibara, 2020-02-19:
αはハードウェアの改良は最小限で，従来手法を使用して**完走を目指す**で良いでしょうか．
ということであれば，GPSは新しい開発になりますので，β（もしかしたらδ）に搭載したほうが良さそうな気がします．

- β(δ)で技術を開発 → αに搭載

という流れが良いと思います．

αで何もしないかというとそうではなくて，αで解決しなければならない問題がいくつか残っていると思います．
例えば以下の問題があると思います．

α固有の問題（改良）
- Rfansの搭載
- 排熱（問題になっているとしたら）
- 外装製作（必要に応じて）
- カメラの取付（改造が必要ですね）

α，β共通
- 信号認識（行うのであれば）
- 探索対象者の認識
- 回路の改良（必要があれば）

taishiyamamoto, 2020-03-15:
設計案です．（あまり #459 と変わらないですが）
カメラは赤いアルミフレームの下に，回路は底面の黄色いスペースの中に取り付ける予定です．
[image]

[image]

taishiyamamoto, 2020-03-17:
αの上の部分に変更を加えず，LiDARの高さを上げる．
雨の日は運用しないようにする．

taishiyamamoto, 2020-06-02:
つくばチャレンジ２０２０が中止になり，現在取り組むことが難しいので，一旦closeします．

closed 2020-06-02 by taishiyamamoto