open-rdc/orne_navigation #459: βのハードウェア開発

tiger0421, 2020-02-18:
### 目的
研究

### 要求
- 配線
- 拡張性

### Sensors
- IMU(ADIS16465)
- VLP16
- web camera or Realsense

yasuohayashibara, 2020-02-19:
こちら，従来のベータとは異なるものになりますので，名称はδというのはいかがでしょうか．
α，β，γときていますので，次の文字を割り当てた方が良さそうな気がします．
もしくはβ2020でしょうか．従来のベータとは異なる名称で読んだほうが良いと思います．

あと，想定されるコンピュータも記載してくれた方が良いです．

taishiyamamoto, 2020-03-01:
設計してみました．
何か問題点があれば指摘して頂ければと思います．

オープンプラットフォームロボットと互換性を持たせたほうが良いと思い，#451 の上にノートPC等を搭載する箱を追加するような設計にしています．
箱の前方に回路やバッテリー等をまとめて配線処理を行い，箱の後方でノートPCを設置することを想定しています．
また，車体前方にカメラ(realsense D435i,T265を想定)を，取っ手にGNSSアンテナを搭載する予定です．
[image]

他の特徴としては，
フリーアングルブラケットを用いて折りたたむことができる．
[image]

箱は前面と後面に2か所ずつで締結されており，取り外しが容易
[image]
[image]

赤い板を外すことでバッテリー交換も可能
（ノートPCをマジックテープ等で固定すれば，恐らく板がなくても安定するので省くことを検討中）
[image]

VLP-16とR-Fans-16両方に対応した取り付け治具を設計
Φ20の半円にLIDAR,非常停止スイッチ，GNSSモジュールの配線を通して運用する予定
[image]

yasuohayashibara, 2020-03-02:
- レバー付きのフリーアングルブラケット良いですね．
- LIDARの位置が低いので，地面を検出するために上に上げた方が良いと思います．
- 重心位置を確認して，転倒しないかを確認してください．
- カメラは雨ざらしにしないほうが良いと思います．（屋根を付けるとか）
直射日光の問題を避けるためにも屋根はあったほうが良いと思います．
- オープンプラットフォームの考え方に従うのであれば，上モノはユニットとして取り外せて，載せてアルファフレームを横からネジで締める程度にしたほうが良いと思います．
あと，電池は即時交換以外は入れたまま充電が前提です．

taishiyamamoto, 2020-03-05:
velodyneのマニュアルにレーザーが飛ぶ領域が記載されていました．
[image]

視野を±15°とすると，現在の設計ではかろうじて視野を確保しているという状態でした．（このページには視野は±10°と記載されていますが）
[image]
[image]

現在のLIDARの位置だと，視野は以下のようになりました．約2.5m先にある地面が検出できます．
[image]

taishiyamamoto, 2020-03-05:
重心位置を求めました．（モバイルバッテリーを2㎏，ノートPCを1.5kgとして計算しました．）
[image]
[image]

taishiyamamoto, 2020-06-02:
下の台車のバージョンが変わったのでそれに対応する必要がある．

taishiyamamoto, 2020-06-16:
バージョンの変化に対応しました．設計コンセプトは上記と変わっていません．
上の箱のCADデータを下記リンクにアップロードします．
https://drive.google.com/drive/folders/1yqpgmu-_5xMmvJX26m-u1H8-Vitg2K3R?usp=sharing

yasuohayashibara, 2020-06-16:
チェックします．

yasuohayashibara, 2020-06-21:
チェックしました．
サイズ感などは良いと思いますが，細部に拘って扱いやすいメカニズムにしたいと思います．
図で示しましたが，全体的なコメントとこだわりを以下列記します．

・orne-boxの天板は上にユニットを乗せる場合には取り外したほうが良いと思います．
・防水にはこだわっているところがあり，できるだけ防水にしたいと考えています．
・防水のためテープとかコーキングで塞ぐのは分解が不要な場所だけにしたいと思います．
・車で移動するときも，分解する場所が少ないものにしたいです．
・排熱ファンは必要ありませんでしょうか．

ロボットを動かしていると，『設計時に対処してくれていればこんなに苦労しないのに』と考えることがあると思います．
アルゴリズムに集中するためには，できるだけ手間のかからないメカニズムを用意することが重要かなと考えています．
その点で，設計に時間をかけても，後の手間を考えるとお釣りがくると思いますので，しっかりと検討して作りたいと思います．

[image]

[image]

[200621_ベータ設計コメント.pptx](https://github.com/open-rdc/orne_navigation/files/4809277/200621_.pptx)

taishiyamamoto, 2020-06-21:
コメントありがとうございます。
指摘事項をチェックボックス化しました。

- [ ] orne-boxの天板は上にユニットを乗せる場合には取り外したほうが良いと思います．

- [ ] 防水にはこだわっているところがあり，できるだけ防水にしたいと考えています．

- [ ] 防水のためテープとかコーキングで塞ぐのは分解が不要な場所だけにしたいと思います．

- [ ] 車で移動するときも，分解する場所が少ないものにしたいです．

- [ ] 排熱ファンは必要ありませんでしょうか．

- [ ] LiDARの配線の検討が必要.

- [ ] LiDARの位置を後ろにしたほうが後ろが検出できる可能がある.

- [ ] 下の樹脂の板に穴をあける必要がある．

- [ ] ネジにラジアル方向の力が加わる．

- [ ] 背面がオープンになっているため，雨天では動かすことができない．

- [ ] 面倒かもしれないが，接合する部品も書いたほうが良い．

- [ ] 上下を配線する穴が無い．

- [ ] orne-boxの天板があるとすると，そちらの対処も必要.

- [ ] 板を固定する穴が少ないかも．
中央でも固定しないと，板が浮いて水が入るおそれがある.

taishiyamamoto, 2020-06-23:
LiDARが高い位置にあると，低層にある障害物を認識できなくなるおそれがあるため，調査する．

taishiyamamoto, 2020-06-23:
Rfansの視野角はマニュアルを見ると以下のようになっていました．
[image]
[image]

taishiyamamoto, 2020-06-23:
現在の設計のLiDARの視野の範囲は以下のようになっています．
地面からLiDARの高さは602mmです．
[image]
[image]

taishiyamamoto, 2020-06-23:
LiDARの地面からの高さと，地面をはじめて検出できる距離をグラフ化したものは以下の通りです．

式は検出距離をy，オフセット（Rfans16の場合は15.8mm）をd，地面からの高さをxとすると，
y = (d + x) / tan(15°)
で算出しています．
[image]

現在の設計(地面からの高さが602mm)では，LiDARの表面からや約2.3mほど離れた地面を検出できることになります．

taishiyamamoto, 2020-06-24:
15.6インチのノートPCであるGALLERIA GCR2070RGF-QCを中に入れた際にどの程度スペースに余裕があるか確認しました．
[image]

現在の設計だと後ろのタブレットとLiDARの視野範囲が干渉していて下げたほうが良いのですが，15.6インチのノートPCが入らなくなりそうです．

ですので，大きい液晶サイズのノートPCを使う場合は小型のモニターを使うなどして基本的に閉じて運用したほうがいいかもしれません．

taishiyamamoto, 2020-06-30:
LiDARとノートPCの位置関係を逆にする案が出たので，それを検討する．

taishiyamamoto, 2021-08-10:
orne_boxのリポジトリができたのでそちらに作業を引き継ぐ．
https://github.com/open-rdc/orne-box

closed 2021-08-10 by taishiyamamoto