open-rdc/orne_navigation #461: αの基板について

shulace13, 2020-02-22:
# 2020/02/19
αの回路をドキュメント化するために基板の調査を行った
## 加えた変更点
- バッテリーからVelodyneとt-frogの電源基板に接続するための基板からUSBを繋げる部分に12Vの出力が二本出ており、使用しないため外した。
（今手元に写真がないためイメージしづらいかもしれないです、すみません）
その他の回路に関しては現状復帰を行った

## 原因
銅製の基板において、グラウンドの確保が不十分であったため、電流が溜まってしまい燃えた
と考えられる。

# 2020/02/21
αにバッテリーを繋いで電源を入れても起動せず銅製の基板が出火した。
グラウンドのパターンが燃えていました。
そこでその基板を撤去し、その他の回路が無事かの確認を安定化電源を用いて確認を行った。
ただしモータの動作確認は行なっていない。
#341 よりリレー回路の撤去も行なった。
緊急停止スイッチはモータのみを遮断するように配線をした。
## 結果
Velodyneは起動及び通信が無事行えた
メイン基板なども無事起動し、パソコン上でデバイスの認識が行えた。

# 今後のToDo

- [ ] バッテリーからVelodyne及び電源基板へ接続するための基板設計・製作

- [ ] 製作した基板を組み込み動作確認を行う

taishiyamamoto, 2020-06-16:
発煙した基板は依然検討されていた充電回路になります． #341
下の画像は充電基板の回路図だと思われます．
[image]

以下の図は，自分が想定している作業のフローチャートになります．
[image]

極端な話，充電システムを使わなければ基板を外して配線処理を行う（回路構成を#341以前に戻す）という対応をとっても自律走行は行えます．
銅製の基板を再現することは必須ではありません．

ryuseiotsuka, 2020-06-25:
αの充電回路を取り除き、配線処理を行いました。
導通を確認し、回路が破損しない事を確認しました。

closed 2020-06-30 by ryuseiotsuka