open-rdc/orne_navigation #470: センサの取り付け治具の作成

taishiyamamoto, 2020-05-26:


taishiyamamoto, 2020-05-26:
センサの取り付け治具を作成する．
作成する治具の一覧を挙げる．

- [ ] realsenseD435i

- [ ] realsenseT265

- [ ] R-fans-16

taishiyamamoto, 2020-06-16:
rfans16とvlp-16を取り付けられるCADモデルを作成しました．
iptファイルとstlファイルをアップロードしておきます．
https://drive.google.com/drive/folders/1xyOZ1PETrPRZbsznUQh_mXcR2zzV6M9d?usp=sharing

@NaokiSato102 
上記CADモデルを3Dプリンタで出力して不具合がないかどうか確認してほしいです．

tiger0421, 2020-07-12:
RealSense D435i用に設計しました．
取り付けるときはRealSense背面のねじ穴を使います．
指が入るように上下に1cmのスペースを付けました．
また，直射日光を遮る効果があるかはわかりませんが，一応屋根を付けました．

- 前から見た図
[image]
- 左から見た図
[image]
- 右から見た図
[image]
- 屋根の長さ
[image]

taishiyamamoto, 2020-07-20:
d435iを取り付けた状態での動作確認
[image]

taishiyamamoto, 2020-07-20:
出力しました．
Rfansの固定のため，長さ25mmのM4の六角穴付きボルトが3本必要になります．
また，治具の固定のため，長さ25mmのM4の六角穴付きボルトが4本必要です．
[image]

yasuohayashibara, 2020-07-20:
林原のネジケースの中にあるかもしれませんが，工場に頼めばいただけけると思います．

taishiyamamoto, 2020-07-23:
@tiger0421 
RealsenseD435iの治具のデータを公開して欲しいです．

tiger0421, 2020-07-24:
.iptと.stlがあればいいよね？
https://github.com/tiger0421/Realsensed435_Case

taishiyamamoto, 2020-07-27:
工作センターに行きましたが、M4×25mmのネジは無いみたいです。

yasuohayashibara, 2020-07-28:
なるほど．もしかしたら持っているかもしれませんが，購入したほうが早いかもしれませんので購入の場合は型番を下さい．

NaokiSato102, 2020-07-28:
先輩作のrealsense T265のデータがiptだったので、Inventorで自前でstlに変換しました。
作成したstl, 3mf, gcodeを先輩のところにアップしておきます。

taishiyamamoto, 2020-07-28:
M4x25mmの六角穴付ボルトの型番です．
https://jp.misumi-ec.com/vona2/detail/110302280360/

@yasuohayashibara 
購入したほうが早そうでしたら4本発注お願いします．

yasuohayashibara, 2020-07-28:
『208 の型番候補があります』と出ます．

taishiyamamoto, 2020-07-28:
すみません．型番を記載します．
CBE4-25

yasuohayashibara, 2020-07-28:
発注しました．

taishiyamamoto, 2020-07-30:
VLP-16を取り付けるためにはインチネジが必要になります．
[image]

この取り付け治具を使う場合，ネジの呼びが1/4inchの長さが約1インチのネジが必要になりそうです．下記にその条件に合いそうな型番を記載しておきます．
CSPPN-SUS-UNC1/4-1

yasuohayashibara, 2020-08-10:
個数を聞こうとして，忘れておりました．
本日３本発注しました．

tiger0421, 2020-08-11:
雑ですがβに取り付けました
[image]

NaokiSato102, 2020-08-20:
realsenseT265の治具の設計に問題あり。
realsenseT265の固定ねじの穴がズレていた。
すでに印刷したものはドリルで穴開けし再利用する。
データは修正を行う。

ネジ頭が飛び出ることに依る問題の発生に備えデータはザクリ穴にしておく。
再利用品でネジ頭問題が出たら再印刷とする。

NaokiSato102, 2020-08-27:
いざ治具に固定しようとしましたがトラブルが発生しました。
T265のネジ穴は[公称M3 0.5 mm pitch](https://www.intelrealsense.com/tracking-camera-t265/#Support)とのことなのですが、
研究室に転がってるM3ネジだと、すっぽ抜けます。
米国製品ですしもしかして忌々しいinchなのでしょうか？
ちょっと検索すると、ネジ穴がバカだとかそういう記述もありましたが、リンク切れしており詳しくは調べられませんでした。

yasuohayashibara, 2020-08-28:
depthが3mmしかありませんので，気をつけて下さい．
ピッチが0.5mmですので，最大6回転です．
もしM3でスッポ抜けるようであれば，既にネジ溝が無くなっているかもしれません．

https://www.intelrealsense.com/wp-content/uploads/2019/09/Intel_RealSense_Tracking_Camera_Datasheet_Rev004_release.pdf?_ga=2.43419244.81405386.1598579520-1878591359.1598579520

NaokiSato102, 2020-08-28:
ともすればすでに死んでおられますね。両面テープ接続に移行しようと思います。強力な両面テープはありますでしょうか？なければ私物がちょうどよいのがありますのでそれを使います。

yasuohayashibara, 2020-08-30:
T265を使用する方が回答お願いします．

NaokiSato102, 2020-09-01:
ミーティングにより両面テープによる固定が承認された。
両面テープ固定による問題が発生した場合、ハードウェア的固定方法を備えた治具を設計する。
強力両面テープは研究室になかったので私物を使う。

固定完了次第、試験運用を行う。
私はジョイスティック操作の方法はうろおぼえながら知っているが、センサ周りの使用方法はわからないので教えていただきたい。

NaokiSato102, 2020-09-17:
私物協力両面テープにより固定の後、適当なアルファフレームへのねじ止めを行った。

- 両面テープによる固定の様子
[image]
- 固定後のケーブルの様子
[image]
[image]
- ねじ止め後
[image]
ネジのクリアランスが狭く、ケーブル接続後にネジを通すのは不可能である。ネジを通した後にケーブルを通し、ねじ止めをするのは可能。差し込み角度的にさせないのでザグリ穴でネジ頭を逃がしても後からネジを入れるのは不可能である。あと、固定ねじの系を今までM4と誤認していたが、M5であったと判明。よって固定用ネジ穴はΦ5.1へ拡張を行った。
[image]
[image]

ネジ穴の系の反映をして、来週のミーティングで確認の後このissueは閉じたいと思う。動作試験は要りますか？

NaokiSato102, 2020-09-17:
3Dモデルへのネジ系の反映とついでにザグリ穴追加をおこなったv3のアップロードを行った。

closed 2020-10-07 by NaokiSato102