open-rdc/orne_navigation #471: VTCのセットアップ

tiger0421, 2020-06-09:
## 環境
- Intel® Core™ i7-9700 CPU @ 3.00GHz × 8 
- NVIDIA RTX 2060 super
- RAM 32GB

## UE4 Editor（version 4.24.3）のインストール
### 所要時間
- makeに約45分

### ビルド後の容量
[image]

## VTCの起動（外部アセットは未ダウンロード）
### エラー内容
UE4Editorから[ここ](https://github.com/furo-org/VTC.git)でクローンしたVTC.uprojectを選択して起動すると次のようなログがでた．

```

```
[image]

その後，「Open with Null Source Code Access」を押し，新しくできたがったVTC.uprojectをUE4Editorで起動しようとしたところ
[image]

「Yes」を選択すると
[image]

tiger0421, 2020-06-09:
## UE4 Editor（version 4.24.3）のインストール
### 所要時間
- makeに約45分

### ビルド後の容量
[image]

Souya25, 2020-06-10:
## VTC.log
Log file open, 06/10/20 12:46:07
LogWindows: Failed to load 'aqProf.dll' (GetLastError=126)
LogWindows: File 'aqProf.dll' does not exist
LogWindows: Failed to load 'VtuneApi.dll' (GetLastError=126)
LogWindows: File 'VtuneApi.dll' does not exist
LogWindows: Failed to load 'VtuneApi32e.dll' (GetLastError=126)
LogWindows: File 'VtuneApi32e.dll' does not exist
LogConsoleResponse: Display: Failed to find resolution value strings in scalability ini. Falling back to default.
LogInit: Display: Running engine for game: VTC
LogPlatformFile: Not using cached read wrapper
LogTaskGraph: Started task graph with 5 named threads and 35 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.087415
LogD3D11RHI: Loaded GFSDK_Aftermath_Lib.x64.dll
LogICUInternationalization: ICU TimeZone Detection - Raw Offset: +9:00, Platform Override: ''
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin AISupport
LogPluginManager: Mounting plugin LightPropagationVolume
LogPluginManager: Mounting plugin ActorLayerUtilities
LogPluginManager: Mounting plugin AnimationSharing
LogPluginManager: Mounting plugin CLionSourceCodeAccess
LogPluginManager: Mounting plugin CodeLiteSourceCodeAccess
LogPluginManager: Mounting plugin GitSourceControl
LogPluginManager: Mounting plugin KDevelopSourceCodeAccess
LogPluginManager: Mounting plugin NullSourceCodeAccess
LogPluginManager: Mounting plugin PerforceSourceControl
LogPluginManager: Mounting plugin PlasticSourceControl
LogPluginManager: Mounting plugin SubversionSourceControl
LogPluginManager: Mounting plugin UObjectPlugin
LogPluginManager: Mounting plugin VisualStudioCodeSourceCodeAccess
LogPluginManager: Mounting plugin VisualStudioSourceCodeAccess
LogPluginManager: Mounting plugin XCodeSourceCodeAccess
LogPluginManager: Mounting plugin AssetManagerEditor
LogPluginManager: Mounting plugin CryptoKeys
LogPluginManager: Mounting plugin CurveEditorTools
LogPluginManager: Mounting plugin DataValidation
LogPluginManager: Mounting plugin EditorScriptingUtilities
LogPluginManager: Mounting plugin FacialAnimation
LogPluginManager: Mounting plugin GameplayTagsEditor
LogPluginManager: Mounting plugin MacGraphicsSwitching
LogPluginManager: Mounting plugin MaterialAnalyzer
LogPluginManager: Mounting plugin MobileLauncherProfileWizard
LogPluginManager: Mounting plugin PluginBrowser
LogPluginManager: Mounting plugin SpeedTreeImporter
LogPluginManager: Mounting plugin DatasmithContent
LogPluginManager: Mounting plugin VariantManagerContent
LogPluginManager: Mounting plugin AlembicImporter
LogPluginManager: Mounting plugin AutomationUtils
LogPluginManager: Mounting plugin BackChannel
LogPluginManager: Mounting plugin CharacterAI
LogPluginManager: Mounting plugin GeometryCache
LogPluginManager: Mounting plugin PlatformCrypto
LogPluginManager: Mounting plugin ProxyLODPlugin
LogPluginManager: Mounting plugin PythonScriptPlugin
LogPluginManager: Mounting plugin SkeletalReduction
LogPluginManager: Mounting plugin MagicLeap
LogPluginManager: Mounting plugin MagicLeapMedia
LogPluginManager: Mounting plugin MLSDK
LogPluginManager: Mounting plugin AndroidMedia
LogPluginManager: Mounting plugin AvfMedia
LogPluginManager: Mounting plugin ImgMedia
LogPluginManager: Mounting plugin MediaCompositing
LogPluginManager: Mounting plugin MediaPlayerEditor
LogPluginManager: Mounting plugin WebMMedia
LogPluginManager: Mounting plugin WmfMedia
LogPluginManager: Mounting plugin TcpMessaging
LogPluginManager: Mounting plugin UdpMessaging
LogPluginManager: Mounting plugin ActorSequence
LogPluginManager: Mounting plugin LevelSequenceEditor
LogPluginManager: Mounting plugin MatineeToLevelSequence
[... 49 lines cut ...]
LogModuleManager: Shutting down and abandoning module PythonScriptPluginPreload (46)
LogModuleManager: Shutting down and abandoning module PlatformCryptoOpenSSL (44)
LogModuleManager: Shutting down and abandoning module PlatformCryptoTypes (42)
LogModuleManager: Shutting down and abandoning module PlatformCrypto (40)
LogModuleManager: Shutting down and abandoning module AnimationModifiers (38)
LogModuleManager: Shutting down and abandoning module PropertyEditor (37)
LogModuleManager: Shutting down and abandoning module AudioEditor (34)
LogModuleManager: Shutting down and abandoning module TextureCompressor (32)
LogModuleManager: Shutting down and abandoning module RenderCore (30)
LogModuleManager: Shutting down and abandoning module Landscape (28)
LogModuleManager: Shutting down and abandoning module SlateRHIRenderer (26)
LogModuleManager: Shutting down and abandoning module OpenGLDrv (24)
LogModuleManager: Shutting down and abandoning module D3D11RHI (22)
LogModuleManager: Shutting down and abandoning module AnimGraphRuntime (20)
LogModuleManager: Shutting down and abandoning module Renderer (18)
LogModuleManager: Shutting down and abandoning module Engine (16)
LogModuleManager: Shutting down and abandoning module CoreUObject (14)
LogModuleManager: Shutting down and abandoning module NetworkFile (12)
LogModuleManager: Shutting down and abandoning module CookedIterativeFile (10)
LogModuleManager: Shutting down and abandoning module StreamingFile (8)
LogModuleManager: Shutting down and abandoning module SandboxFile (6)
LogModuleManager: Shutting down and abandoning module PakFile (4)
LogModuleManager: Shutting down and abandoning module RSA (3)
LogExit: Exiting.
Log file closed, 06/10/20 12:46:16

Souya25, 2020-06-10:
## 環境
windows 10 バージョン2004
Intel Core i7-8700 CPU @3.20GHz
NVIDIA GeForce GTX 1060 6GB
RAM 16GB
## エラー
VTC.uproject を実行
[image]
はいを選択
[image]

## VTC.log

> Log` file open, 06/10/20 12:46:07
> LogWindows: Failed to load 'aqProf.dll' (GetLastError=126)
> LogWindows: File 'aqProf.dll' does not exist
> LogWindows: Failed to load 'VtuneApi.dll' (GetLastError=126) [...]

Souya25, 2020-06-10:
## 新しいプロジェクトを作ってみた
Games
Third Person
C++

> プロジェクトファイルの作成時にエラーが発生しました
> 
> Running C:/UE/UE_4.24/Engine/Binaries/DotNET/UnrealBuildTool.exe  -projectfiles -project="C:/Users/Souya Watanabe/Documents/Unreal Projects/MyProject/MyProject.uproject" -game -rocket -progress
> Discovering modules, targets and source code for project...
> ERROR: Could not find NetFxSDK install dir; this will prevent SwarmInterface from installing.  Install a version of .NET Framework SDK at 4.6.0 or higher.
> 

### .NET Framework 4.8 Developer Pack をinstall
https://dotnet.microsoft.com/download/dotnet-framework/net48

ここからインストールし
もう一度新しいプロジェクトを実行で開けました。

### 再度　ＶTC　を開く
クラッシュした
おそらくメモリ不足（現在16GB)
### クラッシュレポート
```
Fatal error: [File:D:/Build/++UE4/Sync/Engine/Source/Runtime/Core/Private/GenericPlatform/GenericPlatformMemory.cpp] [Line: 186] Ran out of memory allocating 1073741824 bytes with alignment 0

UE4Editor_Core
UE4Editor_Core
UE4Editor_Core
UE4Editor_ImageCore
UE4Editor_ImageCore
UE4Editor_ImageCore
UE4Editor_TextureCompressor
UE4Editor_TextureCompressor
UE4Editor_TextureCompressor
UE4Editor_Engine
UE4Editor_Engine
UE4Editor_Engine
UE4Editor_Engine
UE4Editor_Core
UE4Editor_Core
UE4Editor_Core
kernel32
ntdll

```
### VTC.log
```
Log file open, 06/10/20 17:36:05
LogWindows: Failed to load 'aqProf.dll' (GetLastError=126)
LogWindows: File 'aqProf.dll' does not exist
LogWindows: Failed to load 'VtuneApi.dll' (GetLastError=126)
LogWindows: File 'VtuneApi.dll' does not exist
LogWindows: Failed to load 'VtuneApi32e.dll' (GetLastError=126)
LogWindows: File 'VtuneApi32e.dll' does not exist
LogConsoleResponse: Display: Failed to find resolution value strings in scalability ini. Falling back to default.
LogInit: Display: Running engine for game: VTC
LogPlatformFile: Not using cached read wrapper
LogTaskGraph: Started task graph with 5 named threads and 35 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.088850
LogD3D11RHI: Loaded GFSDK_Aftermath_Lib.x64.dll
LogICUInternationalization: ICU TimeZone Detection - Raw Offset: +9:00, Platform Override: ''
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin AISupport
LogPluginManager: Mounting plugin LightPropagationVolume
LogPluginManager: Mounting plugin ActorLayerUtilities
LogPluginManager: Mounting plugin AnimationSharing
LogPluginManager: Mounting plugin CLionSourceCodeAccess
LogPluginManager: Mounting plugin CodeLiteSourceCodeAccess
LogPluginManager: Mounting plugin GitSourceControl
LogPluginManager: Mounting plugin KDevelopSourceCodeAccess
[... 632 lines cut ...]
[2020.06.10-08.36.13:892][  0]LoadErrors: ロードに失敗しました /Game/Maps/VTC2018/VTC_BuiltData.VTC_BuiltData 参照元 PersistentLevel
[2020.06.10-08.36.14:280][  0]LogShaderCompilers: Display: Worker (1/9): shaders left to compile 2869
[2020.06.10-08.36.14:281][  0]LogTexture: Display: Building textures: RockSmooth0076_1_L (AutoDXT, 2048X2048)
[2020.06.10-08.36.14:281][  0]LogTexture: Display: Building textures: T_TilingRock_A_D (AutoDXT, 2048X2048)
[2020.06.10-08.36.14:907][  0]LogTexture: Display: Building textures: Rock_Formation_Strip_C_D (DXT1, 4096X4096)
[2020.06.10-08.36.14:907][  0]LogTexture: Display: Building textures: T_GDC_StonePath_Tile_M (AutoDXT, 4096X4096)
[2020.06.10-08.36.14:907][  0]LogStaticMesh: スタティックメッシュ HillTree_Tall_02 をビルド中...
[2020.06.10-08.36.18:702][  0]LogTexture: Display: Building textures: Rock_Formation_Strip_C_M (DXT1, 4096X4096)
[2020.06.10-08.36.18:702][  0]LogTexture: Display: Building textures: T_TilingMoss_Clovers-n-Flowers_01_D (AutoDXT, 4096X4096)
[2020.06.10-08.36.18:702][  0]LogTexture: Display: Building textures: Rockface_A_D (DXT1, 4096X4096)
[2020.06.10-08.36.18:703][  0]LogTexture: Display: Building textures: Rockface_A_M (DXT1, 4096X4096)
[2020.06.10-08.36.18:703][  0]LogTexture: Display: Building textures: T_LandscapeMoss_D (AutoDXT, 4096X4096)
[2020.06.10-08.36.18:703][  0]LogShaderCompilers: Display: Worker (6/9): shaders left to compile 2859
[2020.06.10-08.36.18:703][  0]LogTexture: Display: Building textures: T_Tree_Stump_01_N (BC5, 8192X8192)
[2020.06.10-08.36.18:703][  0]LogMemory: Warning: Freeing 33554432 bytes from backup pool to handle out of memory.
[2020.06.10-08.36.18:703][  0]LogMemory: Warning: MemoryStats:
	AvailablePhysical 2540892160
	 AvailableVirtual 140723702378496
	     UsedPhysical 7089483776
	 PeakUsedPhysical 7155294208
	      UsedVirtual 7984242688
	  PeakUsedVirtual 9198829568
[2020.06.10-08.36.18:703][  0]Allocator Stats for TBB: (not implemented)
[2020.06.10-08.36.18:703][  0]LogWindows: Windows GetLastError: この操作を正しく終了しました。 (0)
```

yasuohayashibara, 2020-06-11:
資料を見ると，16GBあれば大丈夫と書かれているものが多いですが，他のアプリケーション（例：chrome）を起動しながらだと不足することがあると書かれていますね．

海外のサイトを見ると，48GBでエラーが無くなったという投稿もありました．

> 再度　ＶTC　を開く
> クラッシュした
> おそらくメモリ不足（現在16GB)

yosagi, 2020-06-15:
DerivedDataCache(DDC)を構築している過程でメモリ不足になっているようですね。DDCは初めてアセットを開くときに構築するデータで、一度構築すれば2回目以降の起動では再利用されます。試したことはないのですが、もしかしてクラッシュしても何度か起動していると少しずつ進捗が進んだりしないでしょうか。
なお、DDCはAppData/Local/UnrealEngineの下に作られます。

tiger0421, 2020-06-24:
アドバイスもいただきましたが，Windows版では，メモリを増設した(16GB -> 32GB)ところエディタが起動したようです．
Linux版は手を付けないで，バイナリ版を試すこととなりましたので，このissueはクローズします．
ありがとうございました．

バイナリ版のセットアップのissueは #472
ROSとの通信は #473

closed 2020-06-24 by tiger0421