open-rdc/orne_navigation #476: VTCで地図用のbagファイル取得

tiger0421, 2020-06-30:


Souya25, 2020-07-03:
## 動作環境
シミュレータ側
OS Windows10 バージョン2004
メモリ 32GB
CPU Intel Core i7-8700

ROS側
OS Ubuntu18.04
メモリ 16GB
CPU Intel Core i5-6300U

## 手順
キーボードを使ってロボットの速度指令を送るパッケージを使って
https://github.com/Souya25/vtc_robo_teleop

以下のルートで地図作成用のbagファイルを取得しました
[image]

## Bagファイル
https://chibakoudai-my.sharepoint.com/:f:/g/personal/s18c1137fp_s_chibakoudai_jp/EjrabR33MW5EqpHUvVlanMkBzjrJ_b2yH-vk9x4HxRu1TQ?e=qC9lHf

ryuichiueda, 2020-07-03:
zipファイルになってますけど、bagファイルって最初から圧縮して記録したり、あとから圧縮したりできます。そうすると圧縮したままrosbagに読み込めます。

↓ご参考にどうぞ。

https://myenigma.hatenablog.com/entry/2016/05/14/170742

ryuichiueda, 2020-07-03:
あれ、zipファイル、空になってます？

Souya25, 2020-07-04:
bagファイルを圧縮してもう一度アップロードしました
https://chibakoudai-my.sharepoint.com/:f:/g/personal/s18c1137fp_s_chibakoudai_jp/EjrabR33MW5EqpHUvVlanMkBzjrJ_b2yH-vk9x4HxRu1TQ?e=1NfmJu

tiger0421, 2020-07-07:
ファイルが見当たりません
[image]

Souya25, 2020-07-07:
[image]
これでいけますか
https://chibakoudai-my.sharepoint.com/:f:/g/personal/s18c1137fp_s_chibakoudai_jp/EjrabR33MW5EqpHUvVlanMkBBw64_t05x77aKmE1iW2P1A?e=I6L0fX

tiger0421, 2020-07-07:
zipは容量が0byteだったので未確認ですが，.bagの方は開けました．
[image]

Souya25, 2020-07-14:
bagファイルの取得と共有ができたのでクローズします

closed 2020-07-14 by Souya25

uhobeike, 2020-09-01:
新たに取り直した
vtcのrosbagデータ(rosbag compressで圧縮済み)

https://1drv.ms/u/s!AomkMtsKOUnSglil7jAdiySMr_cq?e=LnztZM