open-rdc/orne_navigation #479: VTCでmcl-3dl による自律移動

tiger0421, 2020-06-30:


tiger0421, 2020-08-14:
@yugoueda 
遅くなってしまいましたが地図を作成したのでお願いします．
/virtual_tsukuba を使ってください
https://github.com/tiger0421/Tsukuba_MAP

yasuohayashibara, 2020-08-15:
@tiger0421 少々別件になりますが，GitHubに簡単でも良いので地図生成の方法を書いておいてくれると，みなさんの参考になります．よろしくおねがいします．

tiger0421, 2020-08-15:
地図の作り方ですが，以前書いたものに追記しました
https://github.com/open-rdc/orne_navigation/wiki/LeGO-LOAM%E3%81%AE%E9%81%A9%E7%94%A8%E6%96%B9%E6%B3%95

tiger0421, 2020-08-24:
VTCはROS_TIMEにバグがあるかもしれない
mcl_3dlで，VLP16とIMUとodomを使って自己位置推定をしようとしたところ，以下のようなエラーが出た
```
[ INFO] [1598283320.382849798, 1594530257.032875904]: Failed to transform pointcloud: Lookup would require extrapolation into the future.  Requested time 1594530256.283085000 but the latest data is at time 39.037216000, when looking up transform from frame [base_link] to frame [odom]
```

tiger0421, 2020-09-03:
時刻合わせを行い，LIO-SAMで作成した地図を用いて自己位置推定を行おうとした
結果，地面のみがマッチング（動画では白い点で表現）しており，建物などのランドマークを使っていなかった．
https://youtu.be/_Cul8oZWLKo

本来であれば，画像のようになっていてほしい
[image]

---
使用したbagファイル
https://1drv.ms/u/s!AomkMtsKOUnSglil7jAdiySMr_cq?e=LnztZM
_Originally posted by @uhobeike in https://github.com/open-rdc/orne_navigation/issues/491#issuecomment-684246861_

使用した地図
https://drive.google.com/file/d/1Uwfp-8_bIER2_R1xI82eJ2LVPm_ymu9h/view?usp=sharing
_Originally posted by @uhobeike in https://github.com/open-rdc/orne_navigation/issues/491#issuecomment-684636262_

uhobeike, 2020-09-04:
こういうのですか？
[自己位置がずれ始めてきた場所](https://i.gyazo.com/a1004f4370ae4b23ca1cba3f1f5df402.jpg)
[スタート付近](https://i.gyazo.com/c34bee21dc6e52992229f1d0fc28ebb0.jpg)

uhobeike, 2020-09-04:
パラメータ調整で何とか、なりそうな感じですかね？

tiger0421, 2020-09-04:
@uhobeike 
Wikiに[mcl_3dlの適用](https://github.com/open-rdc/orne_navigation/wiki/mcl_3dl%E3%81%AE%E9%81%A9%E7%94%A8)方法を書いてて，[パラメータ](https://github.com/open-rdc/orne_navigation/blob/melodic-devel/orne_navigation_executor/param/3d_localization.yaml)もpushしてあるからこれでやってみてほしい

uhobeike, 2020-09-04:
パラメータ調整しました。
こんな感じでどうですか？
https://www.youtube.com/watch?v=gHUiftKDlPM&t=27s

tiger0421, 2020-09-04:
さっそくありがとう
2つ確認したいんだけど，
1. 黄土色の点（unmatch）が見えなかったけど，解像度の生徒化ではなく実際にほぼすべての点群が白（match）している感じ？
2. パラメータはpushされている奴からさらにチューニングした？

uhobeike, 2020-09-04:
unmatchトピックのデータは全然表示されない感じなので、ほぼすべてのmatchしてるんだと思います。
試しに、unmatchのみにしてどうなるか見てみます。
>黄土色の点（unmatch）が見えなかったけど，解像度の生徒化ではなく実際にほぼすべての点群が白（match）している感じ？
pushされているやつと、実際にmcl_3dlのコードを読んでチューニングしました。(およそ5時間ぐらい)
>パラメータはpushされている奴からさらにチューニングした？

tiger0421, 2020-09-04:
了解！ありがとう

まじでお疲れ様です
>pushされているやつと、実際にmcl_3dlのコードを読んでチューニングしました。(およそ5時間ぐらい)

uhobeike, 2020-09-04:
動画撮りました。やってみたところ、「match9割  unmatch1割」な感じがします。（回転中になることが多い）
（unmatchだと大きな緑の箱が出てきます）
これぐらいのunmatchだと許容って感じですかね？
https://youtu.be/v1vGNTbkfyo
>これ用にまた動画作成します。
>> 試しに、unmatchのみにしてどうなるか見てみます。

チューニングしたパラメータです。そんなにパラメータ数はないです。
あと、これやり過ぎなんじゃないかとか、あったら教えてください。(初めてチューニング(mcl_3dl)したので)
https://github.com/uhobeike/mcl_3dl/blob/master/config/test_localization.yaml

tiger0421, 2020-09-04:
動画を最後まで見てないけどunmatchは問題ないと思う．unmatchが目立ってた4:30~5:10頃のunmatchは地図に登録されていないやつだから．
もし気になるなら地図を作るときに，走行するエリアの外までロボットを走らせて完全（？）な地図を作ればそこのunmatchは減ると思う．
> これぐらいのunmatchだと許容って感じですかね？

パラメータに関してはうまく推定できているように見えるしもう大丈夫だと思うけど，何か気になる場面あった？
このリポジトリにpushされてるパラメータはORNEのために現地のつくば用に調整したパラメータだから無理にそこに合わせなくても大丈夫．
俺もあまりパラメータは詳しくないけど，しいて言うならまだ調査してないodom_errを実験から求めることになるのかな...?
あとは，今回はシミュレータだから問題ないのかもしれないけど，VLP-16の測定距離が0.1~100.0mだから，実際にはclip_nearとclip_farの0.1と100は調整したほうがいいのかも．
> あと、これやり過ぎなんじゃないかとか、あったら教えてください。(初めてチューニング(mcl_3dl)したので)

とりあえず，mcl-3dlを動かせることを確認できたのでこのissueは今度のミーティングでクローズしたいと思います．
もし興味があったらwaypoint ナビゲーション #480 もやってください

uhobeike, 2020-09-05:
なるほど、その手もありますね。
>もし気になるなら地図を作るときに，走行するエリアの外までロボットを走らせて完全（？）な地図を作ればそこのunmatchは減ると思う．

odomの誤差はシュミレータ上だと考慮したほうがいいかなと、ちょっと気掛かりでした。
実験してみるしかなさそうですね。(どうやってやろうかな)
>パラメータに関してはうまく推定できているように見えるしもう大丈夫だと思うけど，何か気になる場面あった？
俺もあまりパラメータは詳しくないけど，しいて言うならまだ調査してないodom_errを実験から求めることになるのかな...?

ギリギリで実際に使わないってことですよね。（調整してみます）
>VLP-16の測定距離が0.1~100.0mだから，実際にはclip_nearとclip_farの0.1と100は調整したほうがいいのかも．

ライトニングトークまでに色々やっておきたいので、月曜日やってみます。(学校に行って)
>もし興味があったらwaypoint ナビゲーション #480 もやってください

tiger0421, 2020-09-11:
自己位置推定できることを確認したのでclose

closed 2020-09-11 by tiger0421