open-rdc/orne_navigation #482: orne_αの故障箇所のチェック

taishiyamamoto, 2020-07-14:
#461 の回路の故障が原因でorne_αが動かなくなった．

本issueではorne_αの回路とモーターの動作テストを行い，故障箇所をチェックする．

taishiyamamoto, 2020-07-14:
故障している可能性のある部品のリストをチェックボックスにします．調査が終わった箇所はチェックしてください．

- [ ] モータードライバ http://t-frog.com/products/motor_driver/

- [ ] 電源基板 http://t-frog.com/products/power_supply/

- [ ] リレー回路 https://github.com/open-rdc/orne_navigation/files/3929867/orne_a_PowerToMotor.pdf

- [ ] モーター http://t-frog.com/products/motor/

他にチェック項目があれば追加してください．

taishiyamamoto, 2020-07-14:
モータードライバの動作テストの動画です．
https://youtu.be/ITeTRMZMI-I

taishiyamamoto, 2020-07-19:
#201 でORNEで主に発生する故障に関して夏迫さんと渡辺先生で議論したものがあったので共有しておきます．
http://t-frog.com/forums/forum.php?ml=tf-2md3-devel&read=183

taishiyamamoto, 2020-07-20:
ORNE-αの電源基板の導通チェックを行いました．
結果を以下にまとめます．

- LCフィルタ用のL200のインダクタが導通していませんでした．

- F200(5V用フューズ),F201(12V用フューズ),F202(TopURG用フューズ)は導通しています．負荷をかけた実験は行っていません．

- ダイオードのD200,D201,D202は順方向には導通し，逆方向には導通しないことを確認しました．

ryuseiotsuka, 2020-07-20:
電源回路の故障診断のページにL200が故障している場合交換するとありますが、部品の型番の記載はどこかにありますか？

ryuseiotsuka, 2020-07-21:
辻電子に回路図を公開してもらえるよう連絡する

ryuseiotsuka, 2020-07-23:
回路の故障を直すため、モータと電源回路、モータードライバの3点をまとめて修理に出す

yasuohayashibara, 2020-07-23:
@ryuseiotsuka のコメントにあるように，リソースを割くことが難しい状況ですので，メーカーに修理を依頼することにしました．
現在，ORNE-α，γの部品（電源，モータドライバ，モータ）は梱包済みです．

@ryuseiotsuka @taishiyamamoto 
故障の経緯を書いたドキュメントを作成して下さい．
（ツジ電子の方が読んで，経緯が分かるようにお願いします．）

ドキュメントを頂いたら，修理に出します．
修理には１～２ヶ月かかりますので，できるだけ早めにいただけると助かります．

ryuseiotsuka, 2020-07-27:
@yasuohayashibara @taishiyamamoto 
ツジ電子に提出するドキュメントを作成しました。確認と訂正をお願いします。
文章の構成上ORNEα,γと見出しをつけているので相手方に提出する際には見出しだけ編集をお願いします。
https://docs.google.com/document/d/1AjmC1y1onf6eoffLD2XKztM6N9DdIiJnsf5vjxK8HlM/edit?usp=sharing

yasuohayashibara, 2020-07-27:
明日，ミーティングですので，ミーティングで確認して修理に出したいと思います．

ryuseiotsuka, 2020-07-28:
送付する文章の確認が終了したのでツジ電子に部品を送付して閉じます

closed 2020-07-28 by ryuseiotsuka

yasuohayashibara, 2020-07-29:
本日(7/29)，ツジ電子に２セット送りました．