open-rdc/orne_navigation #487: データセットの提供

taishiyamamoto, 2020-07-28:


taishiyamamoto, 2020-07-28:
自分たちのサーバーにはつくばチャレンジの実験走行で2015年から2019年に取得したrosbagファイルがあります．
8月1日の説明会を聞き次第，運営に連絡するといった形になると思います．

どのようにしてデータを提供しましょうか．
5年分のbagファイルとなるのでかなりの容量になると思います．

yasuohayashibara, 2020-07-28:
地図生成と自己位置推定に利用できそうな再利用性が高いものをコメント付きで提供するのが良いと思います．
webdavにアクセスしてもらうのも大変だと思いますので，google driveが良いでしょうか．

チームドライブというのが利用できそうです．
https://cloud-work.jp/productivity/google-apps/teamdrive/

taishiyamamoto, 2020-07-29:
チームドライブ作成しました。
https://drive.google.com/drive/u/1/folders/0AMRk5ybblSDcUk9PVA

tiger0421, 2020-07-31:
権限なくてブロックされたのですが，これは念のためにロックを掛けてるのでしょうか？
[image]

yasuohayashibara, 2020-07-31:
閲覧は誰でもできて，書き込みはチームメンバーしかできないという設定が良いと思います．

taishiyamamoto, 2020-08-04:
データセットを提供する際のテンプレート
https://github.com/alexandrx/datasets/tree/alex-restruct-datasets

tiger0421, 2020-09-04:
自分のアカウントにforkしてREADMEに追記した
確認してもらい，問題がなければプルリクを送る
https://github.com/tiger0421/datasets#chiba-institute-of-technology-campus-in-tsudanuma-chiba

tiger0421, 2020-09-11:
マージしていただいた
https://github.com/tsukubachallenge/datasets

closed 2020-09-15 by tiger0421