open-rdc/orne_navigation #491: LIO-SAMを使うために変更するパラメータ

tiger0421, 2020-08-14:
LIO-SAMを使うために変更する必要のあるパラメータを調査する

uhobeike, 2020-08-25:
現状報告
https://i.gyazo.com/dee2d50bf0e3313c8c88a7d82fd25e65.png
以前変わらず、このエラーが出てしまう

uhobeike, 2020-08-25:
現状の動作
https://drive.google.com/file/d/1xSx4IhFbXckbVpZ_TAh3Hv15Dxka2aNi/view?usp=sharing

uhobeike, 2020-08-25:
velodyneのポイントクラウドデータがLIO-SAMの方に上記のエラーが原因で
渡せてないため、mapのフレームが発行されてないと考えている。

uhobeike, 2020-08-27:
windowsとubuntuの時刻同期を行ったところ、出来た。

uhobeike, 2020-08-27:
以下のサイトに方法と結果をまとめた。
https://beike.hatenablog.jp/entry/LIO-SAM/VTC

uhobeike, 2020-08-27:
生成された、PCDデータ
https://youtu.be/xLB_UF2BNso

tiger0421, 2020-08-27:
めちゃくちゃ細かいけどはてなブログの③時刻同期で1行目がwindosになってるのが気になった

uhobeike, 2020-08-27:
指摘ありがとうございます。修正しました。

uhobeike, 2020-09-01:
以下、まとめ終えたブログ

https://beike-re.hatenablog.com/entry/VTC/LIO-SAM/3d-mapping?_ga=2.177404837.1219929223.1598519329-703849559.1590685405

closed 2020-09-01 by uhobeike

uhobeike, 2020-09-01:
vtcのrosbagデータ(rosbag compressで圧縮済み)

https://1drv.ms/u/s!AomkMtsKOUnSglil7jAdiySMr_cq?e=LnztZM

uhobeike, 2020-09-01:
作成したマップ（マージ済み）
https://drive.google.com/file/d/1Uwfp-8_bIER2_R1xI82eJ2LVPm_ymu9h/view?usp=sharing