open-rdc/orne_navigation #492: ORNE-α, γの修理

taishiyamamoto, 2020-08-18:
#482 でツジ電子に送ったモーターと回路が届いた．
そのため，動作確認を行ったのち，αとγに取り付ける．

情報のまとめは #484

taishiyamamoto, 2020-08-24:
#484で発注した駆動輪，モータードライバ，電源基板の動作チェックを行いました．
安定化電源を用いてテストしたところ，特に異常はなく動作しました．

orne-γはジョイスティックコントローラーで動作できるような状態にしています．
https://youtu.be/SL_t7qta-w4

ORNE-αは2接点の非常停止スイッチが故障しているのとリレー回路にサージが発生する問題があるため，まだ動作させることができません．

@yasuohayashibara 
下記に非常停止スイッチの型番を記載したのでよろしければ発注お願いします．
AVN302NR

@ryuseiotsuka 
リレー回路にダイオードの取り付けをお願いします．型番は以下のようなものはどうでしょうか？
http://akizukidenshi.com/catalog/g/gI-00126/

yasuohayashibara, 2020-08-24:
> @yasuohayashibara
> 下記に非常停止スイッチの型番を記載したのでよろしければ発注お願いします．
> AVN302NR

発注しました．

yasuohayashibara, 2020-08-27:
スイッチとダイオードですが共に発送されたとの連絡がありました．
うまくいけば本日中に到着すると思います．

yasuohayashibara, 2020-08-27:
検収室からスイッチとダイオードが到着したとの連絡がありました．

taishiyamamoto, 2020-08-27:
受け取りました

taishiyamamoto, 2020-08-27:
新しい非常停止スイッチをαに取り付けました。

taishiyamamoto, 2020-08-27:
100V1A整流ダイオードをリレー回路に取り付けテストしてみたところ、非常停止スイッチを上げて電源をオンにした時にダイオードが破裂しました。

これは整流ダイオードの耐圧が足りていない可能性はありますか？アノードとカソードを逆に取り付けていることも考えられますが…

yasuohayashibara, 2020-08-28:
以下のダイオード方式と同じ向きに入っているとしたら耐圧の不足です．

https://www.fa.omron.co.jp/guide/faq/detail/faq02804.html

yasuohayashibara, 2020-08-28:
スイッチとコンデンサの納品書を林原までお願いします．

taishiyamamoto, 2020-08-28:
納品書は研究室の自分の机の上にあります．月曜日にデスクの上に置いておきます．

taishiyamamoto, 2020-08-28:
リレー回路のサージに関する調査https://github.com/open-rdc/orne_navigation/issues/455#issuecomment-649234897
調査を見る限り取り付けるとしたらこのあたりでしょうか？
http://akizukidenshi.com/catalog/g/gI-00934/

yasuohayashibara, 2020-08-30:
データを残しておくというのは重要ですね．
#455 の波形を見る限りリレーON時には大きなノイズは発生していないようです．
リレーOFF時にはサージ対策をしていないと-200Vを超える電圧が発生していますね．

まずは，現在のダイオードを取り付けて電圧を計測したほうが良いと思います．

・電流制限のある安定化電源
・オシロスコープ

を使用してまずは挙動を観測したほうが良いです．

大変ですが，これを行ってから次に進めた方が良いと思います．

ryuseiotsuka, 2020-09-03:
安定化電源とオシロを使用して電圧の測定を行いました
### ダイオードなし
<div align="center">
[image]
</div>

### ダイオードあり
<div align="center">
[image]
</div>

*** 

購入したダイオードによってサージ電流を軽減することを確認したため、リレー回路にはんだ付けを行いました
### 半田付けした回路
<div align="center">
[image]
</div>

ryuseiotsuka, 2020-09-03:
原因究明のためα、βの回路ボックスを見比べて確認した所、βのリレー回路の接続が誤っており、繋がっているもののリレーとしての用途は果たしていませんでした。そのためα、γが破損した際にもβは破損しなかったと考えられます。

yasuohayashibara, 2020-09-04:
まず，安定化電源で動作確認してよかったですね．
上記『**半田付けした回路**』の配線が間違っていたという認識で間違いないでしょうか．
もう少し詳しく説明してもらえると助かります．

ryuseiotsuka, 2020-09-04:
恐らく半田付けをしたリレー回路の付近の配線が間違えていると思われます。
まだ原因が詳しくどこにあるかが分かっていない状態かつ、写真等資料が無いと現状を説明しにくいため、月曜日に原因の究明と写真を撮って説明します。

taishiyamamoto, 2020-09-10:
とりあえず問題のあると考えられるリレー回路をORNE-αから取り除き，動かせるような状態にしました．

taishiyamamoto, 2020-09-10:
https://github.com/open-rdc/orne_navigation/issues/492#issuecomment-687026959
> 
> 
> 原因究明のためα、βの回路ボックスを見比べて確認した所、βのリレー回路の接続が誤っており、繋がっているもののリレーとしての用途は果たしていませんでした。そのためα、γが破損した際にもβは破損しなかったと考えられます。

ORNE-βの2020年9月10日時点のシステム図
[image]

リレー回路としての用途を果たしていないことがわかります．
おそらく，つくばチャレンジ2019の実験走行は上記のシステムで動かしていたと考えられます．

taishiyamamoto, 2020-09-10:
ORNE-αのリレー回路の問題点
- b接点の非常停止スイッチではリレー回路の論理が逆になっている．

ORNE-αに搭載されているリレー回路はスイッチを押すとモーターにつながり，スイッチを上げるとセメント抵抗につながります．

ryuseiotsuka, 2020-09-15:
α,γの修理は終了したのでCloseする。

closed 2020-09-15 by ryuseiotsuka