open-rdc/orne_navigation #495: Kinetic環境におけるypspur.hが見つからない問題を解決

NaokiSato102, 2020-09-08:
Kinetic環境において、Orne-navigationを構築しようとするとcatkin build中に以下のエラーが発生する。
`fatal error: ypspur.h: そのようなファイルやディレクトリはありません`

現状の解決法ではbuildが通るまで試行を繰り返す為、これを改良し一発でBuildが通るようにする。

NaokiSato102, 2020-09-08:
そのほか以下のエラーも発生する場合がある。
```
CMake Error at /home/naoki/catkin_ws/src/icart_mini/icart_mini_driver/CMakeLists.txt:18 (find_package):
  By not providing "Findypspur.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ypspur", but
  CMake did not find one.
  
  Could not find a package configuration file provided by "ypspur" (requested version 1.17.0) with any of the following names:
  
    ypspurConfig.cmake
    ypspur-config.cmake

  Add the installation prefix of "ypspur" to CMAKE_PREFIX_PATH or set
  "ypspur_DIR" to a directory containing one of the above files. If "ypspur"
  provides a separate development package or SDK, be sure it has been
  installed.

cd /home/naoki/catkin_ws/build/icart_mini_driver; catkin build --get-env icart_mini_driver | catkin env -si /usr/bin/cmake /home/naoki/catkin_ws/src/icart_mini/icart_mini_driver --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/naoki/catkin_ws/devel/.private/icart_mini_driver -DCMAKE_INSTALL_PREFIX=/home/naoki/catkin_ws/install; cd -
```

試行錯誤の情報を踏まえ、環境構築およびビルド用スクリプトを以下のように改良し、ビルドの高速化を図った。
```
#!/bin/bash

mkdir -p ~/catkin_ws/src
cd ~/catkin_ws/src
catkin_init_workspace
cd ~/catkin_ws/
catkin build
cd ~/catkin_ws/src
git clone https://github.com/DaikiMaekawa/ypspur.git
git clone https://github.com/open-rdc/orne_navigation
wstool init
wstool merge orne_navigation/orne_pkgs.install
wstool up
rosdep install --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y

catkin build icart_mini_driver
sed -i -e 's/find_package(ypspur)/find_package(YPSpur)/g' ~/catkin_ws/src/icart_mini/icart_mini_driver/CMakeLists.txt
catkin build icart_mini_driver
sed -i -e 's/find_package(YPSpur)/find_package(ypspur)/g' ~/catkin_ws/src/icart_mini/icart_mini_driver/CMakeLists.txt
catkin build icart_mini_driver
catkin build icart_mini_driver
catkin build

echo "source ~/catkin_ws/devel/setup.bash" >> ~/.bashrc
exit

```

NaokiSato102, 2020-09-08:
問題解決のためにはicart_mini_driverの上位のフォルダのCMakelistの改良が必要であると示唆された。

NaokiSato102, 2020-09-09:
問題解決に役に立ちそうな検索結果のメモを記しておく
https://www.google.com/search?q=CMakelists+%E9%A0%85%E7%9B%AE
https://ja.wikipedia.org/wiki/CMake
http://sanko-shoko.net/note.php?id=mqz1
https://kamino.hatenablog.com/entry/cmake-list-options
https://www.google.com/search?q=cmake-gui
https://www.google.com/search?q=find_package
https://qiita.com/osamu0329/items/bd3d1e50edf37c277fa9
https://www.google.com/search?q=Workspace+packages+have+changed%2C+please+re-source+setup+files+to+use+them.
https://qiita.com/hyaguchi947d/items/a80844f4a1b14c814326
https://www.google.com/search?q=cmake+CMAKE_PREFIX_PATH
https://kamino.hatenablog.com/entry/cmake_tutorial4
https://stackoverrun.com/ja/q/2065323
https://stackoverrun.com/ja/q/10626669
https://www.google.com/search?q=Add+the+installation+prefix+of+to+CMAKE_PREFIX_PATH+or+set+to+a+directory+containing+one+of+the+above+files.
https://qiita.com/developer-kikikaikai/items/c357706121b3cd9051fe

yasuohayashibara, 2021-01-02:
以下のように変更することでビルドできるようになった．
YPSpurConfig.cmake → ypspurConfig.cmake

https://github.com/open-rdc/ypspur/commit/ac15af6a1cc4f2b5bbd9ab729ee13f412c7d3126

yasuohayashibara, 2021-01-02:
皆さんに確認していただいたらissueを閉じます．

taishiyamamoto, 2021-01-03:
ビルドできました。

yasuohayashibara, 2021-01-03:
@taishiyamamoto 確認ありがとう．
issueは『ypspur.hが見つからない』ということで，これに関しては解決したと思いますのでissueを閉じます．
と思っていたら，対象はKineticでしたね．
(現在melodic-develしか対策していません．)
Kineticに反映させたらissueを閉じます．

NaokiSato102, 2021-01-03:
> ビルドできました。
念のため確認しますが、Kineticの方でビルド確認でしょうか？それともMelodic？もしくは両方？

taishiyamamoto, 2021-01-03:
> > ビルドできました。
> 
> 念のため確認しますが、Kineticの方でビルド確認でしょうか？それともMelodic？もしくは両方？

melodicのみです。

yasuohayashibara, 2021-01-05:
Kineticでもビルドエラーが発生しなくなりましたのでissueを閉じます．

closed 2021-01-05 by yasuohayashibara