open-rdc/orne_navigation #496: ORNEのモーターの電流対策について

ryuseiotsuka, 2020-09-15:
ロボットを手押しした際などに発生する電流の対策について検討する。

taishiyamamoto, 2020-09-19:
つくばチャレンジ2019の技術調査報告でモーター遮断回路について言及されていた．
https://www.slideshare.net/hara-y/tsukuba-challenge-2019-survey
従来はモーター端子短絡ブレーキ＋モーター駆動電源遮断をしていました．
現在はリレー回路を取り除いたので，モーター駆動電源遮断のみになります．
[image]

yasuohayashibara, 2020-09-23:
なるほど．
情報ありがとう．（見たはずなのに忘れていました．）
『モータドライバ駆動電源遮断』は，あまり望ましい方法ではないということですね．
『モータドライバ通信遮断／停止信号』が最も望ましいが，それはつくばチャレンジのルールではNGということですね．

taishiyamamoto, 2020-11-04:
個人的には「モーター端子開放フリー」のみにするのがいいと思います．
以下に使用できそうな部品をまとめておきます．
リレー(3極c接点)：[LY3N DC12](https://jp.misumi-ec.com/vona2/detail/222000863520/?KWSearch=%e3%83%aa%e3%83%ac%e3%83%bc&searchFlow=results2products)
基板用ソケット：[PT11-0 FOR LY](https://jp.misumi-ec.com/vona2/detail/222000863643/?KWSearch=%e3%83%aa%e3%83%ac%e3%83%bc%20%e3%82%bd%e3%82%b1%e3%83%83%e3%83%88&searchFlow=results2products)