open-rdc/orne_navigation #497: wiki内、ORNE環境動作の手順について

masuko-shingo, 2020-10-14:
・自立走行、ナビゲーションのコマンドでwaypointsの記述が足りていないような気がするので実機でのコマンドを残しておきます
wiki上：
roslaunch orne_navigation_executor play_waypoints_nav_{alpha/beta}.launch map_file:=フルパスでmapのyamlファイルを指定
実機が動作したコマンド：
roslaunch orne_navigation_executor play_waypoints_nav_{alpha/beta}.launch map_file:=(フルパスでmapのyamlファイルを指定※拡張子無し) waypoints_file:=(フルパスでwaypointsのyamlファイルを指定)

masuko-shingo, 2020-10-14:
wiki編集しました
issue閉じます

closed 2020-10-14 by masuko-shingo

yasuohayashibara, 2020-10-14:
@masuko-shingo どこに追加したかリンクを張ってもらえればと思います．

masuko-shingo, 2020-10-17:
対応遅れました
すみません
こちらになります
https://github.com/open-rdc/orne_navigation/wiki/ORNE%E7%92%B0%E5%A2%83%E3%81%AE%E6%A7%8B%E7%AF%89%E3%81%AB%E3%81%A4%E3%81%84%E3%81%A6

yasuohayashibara, 2020-10-18:
こちらですね．
了解しました．

https://github.com/open-rdc/orne_navigation/wiki/ORNE%E7%92%B0%E5%A2%83%E5%8B%95%E4%BD%9C%E3%81%AE%E6%89%8B%E9%A0%86