open-rdc/orne_navigation #500: melodicde

yasuohayashibara, 2020-12-05:


yasuohayashibara, 2020-12-05:
icart_miniのgazebo環境を実行したら車輪が回転した．

```
roslaunch icart_mini_gazebo icart_mini.launch
roslaunch icart_mini_driver teleop_joy.launch
```

動画
https://youtu.be/NkI7LKDrjNg

uhobeike, 2020-12-05:
以下の回転しなかった方のrvizの左側にあるタグのrobot modelがエラーになっているのが気になります。
> melodicの環境でgazeboでorne-alphaの車輪が回転しない
>
>動画
>https://youtu.be/gI3OSvjG3ZQ

uhobeike, 2020-12-05:
orne_boxの方は、回転しますか？
[image]

uhobeike, 2020-12-05:
もしかしたら、以下のように設定しているトピックに問題があるのかもしれません。
https://github.com/open-rdc/orne_simulations/blob/afd428b90accccdfeb91a8cd2f174e777b1d8b3c/orne_sim_description/urdf/orne_alpha.gazebo.xacro#L52

https://github.com/open-rdc/orne_simulations/blob/afd428b90accccdfeb91a8cd2f174e777b1d8b3c/orne_sim_description/urdf/orne_alpha.gazebo.xacro#L53

uhobeike, 2020-12-05:
間違ってたら悲しいですが、https://github.com/open-rdc/orne_simulationsの
リポジトリをクローンビルドして試してみました。

使用したコマンド
```
roslaunch orne_gazebo orne_tsudanuma_world.launch `model:=alpha`
rosrun rviz rviz
roslaumch turtlebot3_teleop turtlebot3_teleop_key.launch
```

turtlebotのteleopを使用するために
https://github.com/open-rdc/orne_simulations/blob/afd428b90accccdfeb91a8cd2f174e777b1d8b3c/orne_sim_description/urdf/orne_alpha.gazebo.xacro#L52

https://github.com/open-rdc/orne_simulations/blob/afd428b90accccdfeb91a8cd2f174e777b1d8b3c/orne_sim_description/urdf/orne_alpha.gazebo.xacro#L53

をcmd_velとodomにしました。

実際に操作した所、回転しました。トピックをうまく渡せてないのが問題なのかもしれません。
動画では、うまくrobot modelを読み込めていません、原因は調べてないのでわかりません。（多分気にしなくていいかもしれない）
↓試してみた
https://youtu.be/rfjCFpvuVYg

yasuohayashibara, 2020-12-05:
wikiに書いてある方法で起動したのですが，表示されているロボットのモデルがそもそも違っているように見えます．

https://github.com/open-rdc/orne_navigation/wiki/ORNE%E7%92%B0%E5%A2%83%E5%8B%95%E4%BD%9C%E3%81%AE%E6%89%8B%E9%A0%86

orne_boxは試していませんが，orne_beta, orne_gammaはタイヤが回転しました．

taishiyamamoto, 2020-12-08:
> wikiに書いてある方法で起動したのですが，表示されているロボットのモデルがそもそも違っているように見えます．

orne_navigationとorne_simulationsで，使用しているモデルは異なっています．
大きく異なる点は，差動二輪を表現する際に使用しているgazeboのプラグインになります．

`roslaunch orne_bringup orne_alpha_sim.launch`
使用しているプラグイン：ros_control
https://github.com/open-rdc/orne_navigation/blob/706c9493ad8e0459c761eb523e865a4cadabdcdf/orne_description/urdf/orne_x.gazebo#L6

`roslauch orne_gazebo orne_tsudanuma_world.launch`
使用しているプラグイン：diff_drive_contoller
https://github.com/open-rdc/orne_simulations/blob/afd428b90accccdfeb91a8cd2f174e777b1d8b3c/orne_sim_description/urdf/orne_alpha.gazebo.xacro#L51

yasuohayashibara, 2021-01-04:
原因を追求できていませんが，新しいmelodic-develを使用したら，車輪の回転が見られました．
https://youtu.be/hwyMtRYTNkg

解決しましたので，issueを閉じます．
ご協力いいただいたみなさまありがとうございます．

closed 2021-01-04 by yasuohayashibara