open-rdc/orne_navigation #502: Orne-γにて当方のPCがUTM-30LX、T-flogドライバを認識しなかった。

NaokiSato102, 2020-12-07:
まず、以下のコマンド類を実行する前に試してだめだった。 これは当然。実行する必要がある事を失念していた。
rosrun orne_setup create_robot_params
rosrun orne_setup create_udev_rules

次に、実行後に試したが動かなかった。
UTM-30LXの動作音が電源を入れてもしないので配線を疑ったところ、
なぜかコネクタが外れていた。自然にとれた？

ケーブルを直したところ、ガレリアでは正常動作。当方のPCでは動かないまま。

理由はいくつか考えられる。

- セットアップ不完全
一番有力。完全なセットアップができていない可能性がある。
コマンドの実行洩れがある恐れがある。中途半端に実行されたコマンドの影響を除く手段を教えてほしい。
catkin_wsでcatkin cleanを実行するだけで問題ないか？

- multi_view_imageを構成するアプリケーションとの干渉

- USB周りの不調

対処について情報をいただきたい。

tiger0421, 2020-12-07:
よくわかりませんが，まずはusbが認識されているかどうか確認してください
`
$ ls /dev
`
とかの情報がないとわからないです

NaokiSato102, 2020-12-07:
とりあえずlsusbには載っていませんでした。正常時の挙動を覚えてないのでそもそも載るものかは知りませんが。

taishiyamamoto, 2020-12-08:
ドキュメントを更新しました．
よろしければ参考にしてください．
https://github.com/open-rdc/orne_navigation/wiki/ORNE環境動作の手順

tiger0421, 2020-12-08:
lsusbで出ないなら認識してないです

NaokiSato102, 2020-12-08:
対処はどうしましょうか

tiger0421, 2020-12-08:
usbハブを繋がずに接続してみてください

NaokiSato102, 2020-12-08:
失礼ながら書き忘れていましたが直刺しでもダメでした。

closed 2021-01-26 by tiger0421