open-rdc/orne_navigation #503: orne_navigation内brunch及び内部ファイルが複雑になりすぎている

haruyama8940, 2020-12-11:
brunchが混沌とし、また切り替えたbrunch内に明らかな不要なファイルが存在している

uhobeike, 2020-12-11:
コメントはそれで問題ないけど、issue名は、「orne_navigation内brunchの整理（γ）」で良いかも

haruyama8940, 2020-12-11:
ありがとうございます。修正しました。

taishiyamamoto, 2020-12-28:
自分が関わっている中で不要なブランチは消しました。

yasuohayashibara, 2021-01-02:
以下，整理の第一弾です．
gazeboで動作することは確認しています．
（自律移動のところでウェイポイント上で停止する様子が見られますので，何か問題がありそうですが．）
- orne_navigation (環境構築手順の簡略化)
https://github.com/open-rdc/orne_navigation/tree/brushup
- ypspur (ビルドの問題を解決)
https://github.com/open-rdc/ypspur
- icart_mini
https://github.com/open-rdc/icart/tree/brushup
通常の環境構築をするとmelodic-develが選択されるが，現在は一時的に上記のブランチに切り替える．

yasuohayashibara, 2021-01-03:
kinetic-develに関しても整理しました．
- orne_navigation (環境構築手順の簡略化)
https://github.com/open-rdc/orne_navigation/tree/develop/kinetic-devel
- icart_mini (ビルドの問題を解決)
https://github.com/open-rdc/icart/tree/develop/kinetic-devel

yasuohayashibara, 2021-01-03:
次に，しばらく変更がなくて，これから利用することがなさそうなbranchを整理します．
これらのbranchに関わっている方はマージするか，削除していただければと思います．

https://github.com/open-rdc/icart/branches/stale
https://github.com/open-rdc/orne_navigation/branches/stale

あと，こちらも不要だと思いますので，チュートリアルで使用しそうなもの以外は削除する予定です．
https://github.com/open-rdc/orne_navigation/tree/kinetic-devel/orne_navigation_executor/waypoints_cfg

yasuohayashibara, 2021-01-06:
とりあえず一段落しました．
@uhobeike 先走ってしまいましたが，何かありましたら追加でお願いします．

closed 2021-01-19 by taishiyamamoto