open-rdc/orne_navigation #504: パッケージの整理

taishiyamamoto, 2020-12-28:
現在、あまり使用していないパッケージがワークスペースにインストールされるようになっているので整理する。

taishiyamamoto, 2020-12-28:
[melodic](https://github.com/open-rdc/orne_navigation/tree/melodic-devel)でのインストールされるパッケージは以下のようになっている。
https://github.com/open-rdc/orne_navigation/blob/melodic-devel/orne_pkgs.install

この中で、あまり使用されなくなったパッケージとソースビルドする必要のないパッケージを挙げる。

- あまり使用されなくなったパッケージ
  - [adis_imu_drivers](https://github.com/open-rdc/adis_imu_drivers)　
  - [fulanghua_navigation](https://github.com/open-rdc/fulanghua_navigation)　（waypoint_navのみ使用）
  - [mpu6050_serial_to_imu](https://github.com/Nasupl/mpu6050_serial_to_imu)　
  - [hokuyo3d](https://github.com/at-wat/hokuyo3d)
- ソースビルドする必要のないパッケージ(aptでインストールできるようになったパッケージ)
  - [openslam_gmapping](https://github.com/ros-perception/openslam_gmapping)
  - [slam_gmapping](https://github.com/ros-perception/slam_gmapping)
  - [geometry2](https://github.com/ros/geometry2)

taishiyamamoto, 2020-12-28:
方針
- fulanghua_navigationの[fulanghua_waypoints_nav](https://github.com/open-rdc/fulanghua_navigation/tree/melodic-devel/fulanghua_waypoints_nav)をorne_navigationに移動
- [orne_pkgs.install](https://github.com/open-rdc/orne_navigation/blob/melodic-devel/orne_pkgs.install)の整理

taishiyamamoto, 2020-12-28:
orne_pkgs.installを整理したプルリクを投げました。
https://github.com/open-rdc/orne_navigation/pull/505

仮想環境でビルドが通ることを確認したので、多分大丈夫だと思いますが一応確認お願いします。

yasuohayashibara, 2020-12-29:
G

```
[ WARN] [1609203283.607558697, 1352.313000000]: Failed to transform the goal pose from /map into the map frame: Invalid argument "/map" passed to lookupTransform argument source_frame in tf2 frame_ids cannot start with a '/' like: 
[ERROR] [1609203283.607649098, 1352.313000000]: The goal pose passed to this planner must be in the map frame.  It is instead in the /map frame.
[ERROR] [1609203283.607723471, 1352.313000000]: The goal pose passed to this planner must be in the map frame.  It is instead in the /map frame.
Warning: Invalid argument "/map" passed to canTransform argument source_frame in tf2 frame_ids cannot start with a '/' like: 
         at line 134 in /home/mirai/catkin_ws/src/geometry2/tf2/src/buffer_core.cpp
```

closed 2020-12-29 by yasuohayashibara