open-rdc/orne_navigation #509: kineticからmelodicへの変更点

yasuohayashibara, 2021-01-05:
#508 で議論してきたが，色々と気になるところが出てきたので，こちらで議論します．

yasuohayashibara, 2021-01-05:
気になる変更点としては，以下がある．
icart-middleの最大速度を超えていないかチェックする．

https://github.com/open-rdc/orne_navigation/blob/583a0007526b31d19c01d021d7c9b47170050c62/orne_navigation_executor/param/base_local_planner_params_alpha.yaml#L3-L12

 max_vel_x: 0.9 
ロボットの仕様
TF-M30-24-3500-G30L/R
無負荷回転数　　3500rpm
減速比　1/30
ホイールの直径　0.3m
トレッド　0.4m

無負荷回転数を基準に考えると，
最大速度　1.8m/s

軌道計画で要求される最大速度
0.9 + 1.6 * 0.2 = 1.22m/s

トルクに関しては検討していないが，速度は問題なさそうである．

yasuohayashibara, 2021-01-05:
cost_scaling_factorが１になっています．

https://github.com/open-rdc/orne_navigation/blob/583a0007526b31d19c01d021d7c9b47170050c62/orne_navigation_executor/param/costmap_common_params.yaml#L5

こちらの値は距離に対するコストの減少の比率になりますので，小さいとコストがより広がるようです．

https://wiki.ros.org/costmap_2d/hydro/inflation

yasuohayashibara, 2021-01-05:
膨張リセット法のパラメーが変更されています．
たしか保科くんが最適値を検証していたような気がします．

localization_params.yaml

kinetic
reset_th_alpha: 0.003

meolodic
reset_th_alpha: 1.3

yasuohayashibara, 2021-01-05:
@taishiyamamoto 
kinetic-develとmelodic-develのブランチで動きに関与しそうな部分でパラメータの違いが見られます．
OSを変えることで，動きが変わるのは混乱が生じるような気がしますので，どちらかの値に統一したいと思います．
見る限りmelodic-develの方が攻めたパラメータのような気がしますので，最適化はされていなくても今まで安定して動作してきたkinetic-develの方にできるだけ近づけたいと思いますが，いかがでしょうか．
3dとついている部分はmelodic-devel固有の部分だと思いますので変更しない予定です．
ご意見をいただければと思います．

yasuohayashibara, 2021-01-05:
orne_recovery_behaviorを削除しているようですが，戻しても良いでしょうか？

taishiyamamoto, 2021-01-05:
> 膨張リセット法のパラメーが変更されています．
> たしか保科くんが最適値を検証していたような気がします．
> 
> localization_params.yaml
> 
> kinetic
> reset_th_alpha: 0.003
> 
> meolodic
> reset_th_alpha: 1.3

パラメータが大きく異なっている理由を調べたところ、リセットの閾値αの計算方法がkineticとmelodicで異なっていました。

- kinetic パーティクルの重みの2乗和
- melodic パーティクルの重みの和

taishiyamamoto, 2021-01-05:
> @taishiyamamoto
> kinetic-develとmelodic-develのブランチで動きに関与しそうな部分でパラメータの違いが見られます．
> OSを変えることで，動きが変わるのは混乱が生じるような気がしますので，どちらかの値に統一したいと思います． [...]

パラメータを変更した動機は確かこのissueになります。
https://github.com/open-rdc/orne_navigation/issues/481

つくばチャレンジの実験走行を考慮して、自律走行の速度を上げた際にパラメータを変更しました。

yasuohayashibara, 2021-01-05:
賛成します．

> * 自律走行の速度はkinetic,melodic両方上げた状態にする(max_vel_x: 0.9)
なるほど．速度を上げた経緯を忘れていました．
つくばチャレンジにおいては必要な変更ですね．
ただし，屋内の実験では速度が速すぎるかもしれませんので，速度の変更方法をwikiに書くなどして用途に応じて速度を変更したほうが良いかもしれません．
速度を上げると自律移動の難易度が増しますので，みんなで検証しながら最適解を探っていったほうが良いかなと思います．

> * 速度以外の自律走行に関わるパラメータは一旦kineticに合わせる(cost_scaling_factorなど)
こちらは林原が検証しながら行いたいと思います．
基本的にkineticに揃えるだけですので，シミュレータで動作確認したらマージします．
もし実ロボットでより良いパラメータを見つけたらkinetic-develにも反映させたいと思います．
（混在していると少々大変ですね．）

> * 膨張リセットのパラメータはkineticのものと同じにする(reset_th_alpha: 0.003)
こちらはお願いします．

yasuohayashibara, 2021-01-05:
方針が定まったので，情報共有できた頃にissueを閉じます．

yasuohayashibara, 2021-01-05:
@taishiyamamoto 以下，kineticとmelodicで気になる差分です．
もし事情が分かりましたら教えていただければ助かります．

```
orne_bringup/launch/orne_alpha.launch

kinetic
    <remap from="/scan" to="/diag_scan" />
    <param name="frame_id" value="diagonally_hokuyo_link" />
    <param name="serial_port" value="/dev/sensors/hokuyo_H1110848"/>

melodic
    <param name="frame_id" value="hokuyo_link" />	    <param name="frame_id" value="hokuyo_link" />
    <param name="serial_port" value="/dev/sensors/hokuyo_H1110843"/>	    <param name="serial_port" value="/dev/sensors/hokuyo_H1110843"/>
    <param name="publish_multiecho" value="false"/>	    <param name="publish_multiecho" value="false"/>
```
```
orne_description/urdf/sensors/vlp16.urdf.xacro

kinetic
              <samples>720</samples>

melodic
              <samples>721</samples>
```

以下で確認できます．
https://github.com/open-rdc/orne_navigation/pull/510/files

taishiyamamoto, 2021-01-06:
> @taishiyamamoto 以下，kineticとmelodicで気になる差分です．
> もし事情が分かりましたら教えていただければ助かります．

自分のわかる範囲ですが

- orne_bringup/launch/orne_alpha.launch
  - このlaunchファイルのurg_nodeは過去に斜めづけのTOP-URGを搭載していて、それを起動するためのノードだと思います。このissueを見る限り、velodyneを搭載した際にコメントアウトされたようです。https://github.com/open-rdc/orne_navigation/issues/358#issuecomment-414544597 そして、自分はそれを水平に取り付けられているTOP-URGを起動させるように書き換えました。ですが、以下で水平に取り付けられたURGは起動できるようになっているので重複しており、この変更は間違っていると思います。
```
    <include file="$(find icart_mini_driver)/launch/icart_mini_drive.launch">
        <arg name="model"    value="$(find xacro)/xacro.py '$(find orne_description)/urdf/orne_alpha.xacro'"/>
        <arg name="scan_dev" value="/dev/sensors/hokuyo_H1110843"/>
		<arg name="ypspur_params" value="/usr/local/share/robot-params/orne_x.param"/>
    </include>
```
- orne_description/urdf/sensors/vlp16.urdf.xacro
  - 申し訳ありませんがよく覚えていません。（事情は聞いたはずなのですが）島田君に聞けばわかるかもしれないです。

tiger0421, 2021-01-06:
>  ```
> orne_description/urdf/sensors/vlp16.urdf.xacro
> 
> kinetic
>               <samples>720</samples>
> 
> melodic
>               <samples>721</samples>
> ```

これはvlp16のscan（2次元）の点群数が，実機だと721だったのに対してgazeboだと720しか得られなかったため変更しました．

yasuohayashibara, 2021-01-06:
@taishiyamamoto 調査ありがとう．

orne_bringup/launch/orne_alpha.launch に関して
kineic or melodicで動作実績のある方を知っていたら教えてください．
雰囲気的にmelodicの方かなとは思いますが念のため教えていただければと思います．

taishiyamamoto, 2021-01-06:
最近まで動かしていたのはmelodicになります。

yasuohayashibara, 2021-01-06:
orne_bringup/launch/orne_alpha.launchは #523 で対処しました．
膨張リセット法の変更が残っていると思いますが，それは別途行うこととしてこちらのissueは閉じます．

ちなみにブランチ間の差分は以下のコマンドでチェックすることができます．
フォルダ内，ファイル内の差分など色々とオプションがあります．
（林原ははじめて知ったのですが，今回の件では大活躍でした．）

```
git diff origin/melodic-devel origin/kinetic-devel
```

closed 2021-01-06 by yasuohayashibara