open-rdc/orne_navigation #528: 鉛蓄電池のストックが少ないこと

haruyama8940, 2021-11-07:
現在主にorne-αで用いている鉛蓄電池(LONG WP12-12 )において，正常に使用または長時間使用（例としてつくばチャレンジのコースを完走）できる組，個数が少ない問題があるため，調査を行う．
以下に調査した個数を追加する．

yasuohayashibara, 2021-11-08:
つくばチャレンジの本走行に間に合わせるためには，早めに調査したほうが良いと思います．

haruyama8940, 2021-11-09:
調査しました. 全体数は10個です．
また樋高くんが行った実験で測定した容量は次のとおりです．

使用可能×8
7978 , 7166, 11232, 10615, 9524, 3908, 4275, 4743 [mAh]
使用不可×2

バランスと容量の関係でつくばチャレンジ本番で使用できる組は[7978 , 7166] [11232, 10615] 2つです．

yasuohayashibara, 2021-11-09:
不要なバッテリを廃棄することを条件に新しいバッテリを購入します．
LONG WP12-12　をいくつ必要でしょうか？

haruyama8940, 2021-11-09:
把握しました。
3908, 4275, 4743の寿命が近いと予想されるため、
4つの追加を考えていますが，他に意見ある方はいますか？

yasuohayashibara, 2021-11-10:
購入しようと調べたら２種類あります．
F1端子で良いでしょうか．
サイズが２種類あるようですので，調査して回答してください．

完全密封型鉛蓄電池　１２Ｖ１２Ａｈ　ＷＰ１２－１２（Ｆ１端子）
https://akizukidenshi.com/catalog/g/gB-00066/

haruyama8940, 2021-11-10:
購入数は４つでお願いします．
在庫があるサイトを調査しました．bench.prieure.xyzの方が7000円ほど安いです．
[amazon]
https://www.amazon.co.jp/WP12-12-%E3%80%90%E3%81%8A%E5%BE%97%EF%BC%814%E5%80%8B%E3%82%BB%E3%83%83%E3%83%88%E3%80%91-%E3%80%8A%EF%BC%AC%EF%BC%AF%EF%BC%AE%EF%BC%A7-%E6%80%A7%E8%83%BD%E5%A4%9A%E7%94%A8%E9%80%94%E3%83%90%E3%83%83%E3%83%86%E3%83%AA%E3%83%BC-UPS%E7%94%A8%E3%83%90%E3%83%83%E3%83%86%E3%83%AA%E3%83%BC-%E9%9B%BB%E5%8B%95%E3%83%AA%E3%83%BC%E3%83%AB%E7%AD%89%E3%81%AB%EF%BC%81/dp/B076BCS53P/ref=sr_1_6?adgrpid=104559455997&gclid=CjwKCAiA1aiMBhAUEiwACw25Mdcx6ztukuKSWnjtR9yY6yS72hteze-9oDrW1xlb7-HHftRsnnK3txoCcTYQAvD_BwE&hvadid=553852661463&hvdev=c&hvlocphy=1009250&hvnetw=g&hvqmt=e&hvrand=15166821910716126869&hvtargid=kwd-13314322654&hydadcr=27300_14478774&jp-ad-ap=0&keywords=wp12-12&qid=1636526225&sr=8-6

[bench.prieure.xyz]
https://bench.prieure.xyz/index.php?main_page=product_info&products_id=6003&zenid=q5q0kp9brpo8s2vg637htplvu7

yasuohayashibara, 2021-11-10:
amazonで購入しました．

haruyama8940, 2021-11-10:
ありがとうございます．
議論が終了したと思うため，issueを閉じます

closed 2021-11-10 by haruyama8940