open-rdc/orne_navigation #531: 現在のorne_navigationの整理，不具合の改善をする

haruyama8940, 2022-04-11:
現在のorne_navigationの不具合の洗い出しと必要なパッケージの選択を行う．
調査しながら追記．
対象はnoeticを想定．

uhobeike, 2022-04-12:
パラメータは機体ごとにファルダ分けしたほうが良さそう。
共通なやつは、分けないでおくか、commonフォルダにまとめる。

uhobeike, 2022-04-18:
turtlebot3だとロボットごとにフォルダは分けておらず、環境変数として
`burger, waffle, waffle_pi`のどれかをセットしておくことでファイルの呼び出しを行っています。
https://github.com/ROBOTIS-GIT/turtlebot3/tree/master/turtlebot3_navigation/param

https://github.com/ROBOTIS-GIT/turtlebot3/blob/66681b33749c44e7d9022253ac210ef2da7843a0/turtlebot3_navigation/launch/move_base.launch#L3

https://github.com/ROBOTIS-GIT/turtlebot3/blob/66681b33749c44e7d9022253ac210ef2da7843a0/turtlebot3_navigation/launch/move_base.launch#L11-L12

> 探したのですが，複数のロボットを一つのパッケージで扱うサンプルが見つかりませんでしたので，できればメジャーなパッケージがどのような構造を選択しているかが分かる例があればお願いします．

前例やメジャーなパッケージに従うことは良いと思うのですが、
手の入れやすさや管理のことを考えるとフォルダごとに分けるのはありなのかなと思っています。

私は普段orne_navigationを使用しているわけではないため、あんまり言える立場では無いのですが
使用している林原研の方々はファルダごとに分けたほうが開発がしやすいのか気になります。

masakifujiwara1, 2022-04-18:
現状, param内の共通なものでもロボットごとに変更することがあります. 
共通のものも含めて, ロボットごとにフォルダが分かれている方が扱いやすいです.

haruyama8940, 2022-05-17:
ブランチを変えて作業します

uhobeike, 2022-05-17:
> ブランチを変えて作業します

https://github.com/open-rdc/orne_navigation/tree/refactor/melodic-devel

haruyama8940, 2022-05-17:
ファイルの整理中です
改善
・base_footprintを追加する
・デフォルトだとlocal_costmapが機能しない
・alphaのurgの設定がドキュメント化されてない
・urdfの見直し

haruyama8940, 2022-09-30:
ある程度ファイルの整理を行いました
https://github.com/open-rdc/orne_navigation/tree/refactor/noetic-devel

yasuohayashibara, 2022-10-08:
こちらPRを投げていただき，差分を見ながら議論したほうが良いと思います．
あと小分けにして優先度の高いものからマージしたほうが良いと思います．
あと，個別には話しましたが，iCartをできるだけ利用したほうが良いと思います．
問題があるようでしたら林原の方でも調査しますので，issueを立てていただければと思います．

haruyama8940, 2022-12-15:
こちらの作業は冬休みを利用して議論，作業したいと考えています．