open-rdc/orne_navigation #536: orne_navigationを用いたロボットに発生するwaypoint間のpath planningにおける問題

hellocit, 2023-03-18:
問題が発生したハードウェア：orne-box2
問題が発生した場所：1号館の周辺
問題が発生した具体的な場所は写真の赤丸の位置で発生した．
[image]

hellocit, 2023-03-18:
発生した問題
赤丸の位置で謎の走行を繰り返した．この問題が起こりやすい場所としては，waypoint間の距離が遠いことがあげられる．

yasuohayashibara, 2023-03-19:
> 謎の走行

slackではもっと具体的に書いていたと思いますので，こちらにも情報を追加してください．

---
おそらく以下のことが関連していそうな気がします．

知っていてほしいこととして，

・現在のシステムでは，ロボットは『ウェイポイントをつなぐ線上を移動しません』

１秒間に１回パスプランニングをし直していると思いますが，常に自分の位置を基準にゴールに向かおうとします．
例えば，まっすぐ進もうと思っても右に曲がるロボットは右に曲がった場所からパスプランニングします．
つまり，左に戻ろうとせずに，自分が動いた右の位置を基準としてパスプランニングするので，どんどん右にずれます．
これを解決するためには，頻繁なパスプランニングを行わないようにするか，ロボットがまっすぐに走れるように調整することです．

----
もう一つ分かったこととして，IMUを調整してください．
全ての実験をやり直しになると思いますので，まずはIMUを調整したほう良いと思います．

hellocit, 2023-03-19:
補足説明（謎の走行）
赤丸の中の左のwaypointについて，waypointの方向に向かわず，waypointの手前で右側にそれていった．その後，植木のコストに入って緊急停止ボタンを押し，手動でロボットをwaypoint手前に移動させ，自己位置を修正した後に動作を再開させた．

waypointを通過した後，次のwaypointではなく，１号館側に向かって走行し始めた．その後，１号館を右折するまで，１号館の壁側をギリギリで走行した．6号館側の1号館入り口付近では，１号館の中に入っていく挙動をみせぶつかりそうになったため，緊急停止ボタンを押した．

赤丸の中の右2つのwaypoint付近では，4~6人の集団が前から歩いてきて，ロボットの3〜5ｍ前を横切っていった．

closed 2024-05-31 by hellocit